espruino/BangleApps #2067: New buzz patterns still do not work

nxdefiant, 2022-07-28:
The buzz patterns introduced in #1889 still do not work on my watch (sched v0.13, alarm v0.33, firmware 2v14), e.g. if I select ':' as buzz pattern I only get silence. '.' still is the strong pattern. Does anyone else have this problem?

I have attached the sched.js:
[sched.js.txt](https://github.com/espruino/BangleApps/files/9214012/sched.js.txt)

The code for ":"-pattern seems to be there. Does anyone have a clue what might be wrong?

gfwilliams, 2022-07-29:
I've been using this for a while it seems to work fine for me.

Have you tried actually just calling `require("buzz").buzz(":")` from the IDE and seeing what happens?

nxdefiant, 2022-07-29:
Same, with `require("buzz").pattern(":")` in the console (left) part of the IDE, nothing happens, "." works. I wonder what actually gets executed when I put that in the console, I don't see it fetching buzz.js.

When I paste the full code from buzz.js I get a noise, so it generally works:
```
pattern = pattern => new Promise(resolve => {
  function doBuzz() {
    if (pattern == "") resolve();
    var c = pattern[0];
    pattern = pattern.substr(1);
    const BUZZ_WEAK = 0.25, BUZZ_STRONG = 1;
    const SHORT_MS = 100, MEDIUM_MS = 200, LONG_MS = 500;    
    if (c == ".") Bangle.buzz(SHORT_MS, BUZZ_WEAK).then(() => setTimeout(doBuzz, 100));
    else if (c == ",") Bangle.buzz(MEDIUM_MS, BUZZ_WEAK).then(() => setTimeout(doBuzz, 100));
    else if (c == "-") Bangle.buzz(LONG_MS, BUZZ_WEAK).then(() => setTimeout(doBuzz, 100));
    else if (c == ":") Bangle.buzz(SHORT_MS, BUZZ_STRONG).then(() => setTimeout(doBuzz, 100));
    else if (c == ";") Bangle.buzz(MEDIUM_MS, BUZZ_STRONG).then(() => setTimeout(doBuzz, 100));
    else if (c == "=") Bangle.buzz(LONG_MS, BUZZ_STRONG).then(() => setTimeout(doBuzz, 100));
    else setTimeout(doBuzz, 100);
  }
  doBuzz();
});

pattern(":");
```

nxdefiant, 2022-07-29:
I might have an idea now. I have a widget widshipbell installed that also loads buzz.js. It was installed before the changes to buzz.js, so does not include the new patterns. Now what does happen when multiple files require a module? I guess only the first one wins? So if the widget requires buzz.js and loads the old buzz.js the new buzz.js from the scheduler is not loaded?

If so we maybe should bump the version number on all apps that require a module if that gets updated.

closed 2022-07-29 by nxdefiant

gfwilliams, 2022-07-29:
> Now what does happen when multiple files require a module?

I'd never considered that! Yes, it's the first one that wins.

This wouldn't be an issue for apps, it's just where you have widgets/boot code as those are loaded at the same time as others.

I guess there's a good argument that actually the App Loader should load `buzz.js` into a file called `buzz` in storage - so then all apps use the same version. Also not having that code inside them would make them load quicker