espruino/BangleApps #2068: Backup fails with New interpreter error: MEMORY

nxdefiant, 2022-07-31:
This is BangleJS2 on firmware version 2v14.

The backup from the App loader fails while backing up recorder.log2.csv from the recorder app. In the browser console the last thing I see is:
> <COMMS> readStorageFile "recorder.log2.csv"
comms.js:444 <COMMS> readTextBlock read started...
comms.js:430 <COMMS> size is 130382

Nothing happens after that, I waited for for up an hour.

When I open a parallel connection in https://www.espruino.com/ide/ I see:
> Execution Interrupted
at line 1 col 100
...getLength());var l=f.readLine();while(l!==undefined){Bluetooth.print(l);l=f.r...
                                               ^
in function called from line 7 col 8
    })()
       ^
New interpreter error: MEMORY

The backup from recorder.log2.csv worked before because it was part of the backup I restored a few days earlier.

nxdefiant, 2022-07-31:
Update: I also now get a
> New interpreter error: LOW_MEMORY,MEMORY

while viewing the track in the recorder app. That also worked before.

gfwilliams, 2022-08-01:
Do you have any success reading the file with the Web IDE?

What was showing on your watch's screen during the backup? Was it still showing the watch?

MEMORY/LOW_MEMORY are errors you get when there's not enough memory on the watch. I think that's probably one of two things:

* Actual low memory - you could check by running `process.memory()` from the Web IDE when running the watch normally
* The `recorder.log2.csv` file may have *really long lines* so `f.readLine()` uses up all available memory reading the lines

> The backup from recorder.log2.csv worked before because it was part of the backup I restored a few days earlier.

Do you still have that backup file you could share (even privately?). It's *possible* that the issue is that the **last** backup actually ended up removing all the newlines from that log file, so then after the restore the next backup had trouble (as would the track viewer)

nxdefiant, 2022-08-01:
ah right, I can read the file in the IDE, unfortunately something broke:
[image]
These are the byte values of the content this csv usually starts with, for comparison this is a newly generated file:
[image]

So something broke with backup/restore. I still have the original recorder.log2.csv. Is there a way to upload it as storage file from the web ide?

gfwilliams, 2022-08-01:
> Is there a way to upload it as storage file from the web ide?

I'm not sure... But can you open the  original recorder.log2.csv (from the backup?) and see if that has any newlines in it?

nxdefiant, 2022-08-01:
Yes, there are \n-newlines:
> head -n 3 /tmp/test/storage-files/recorder.log2.csv
Time,Latitude,Longitude,Altitude
1654417218,,,
1654417228,,,

gfwilliams, 2022-08-01:
Is that the recording file from the backup? Or was it one that you downloaded separately at another time?

nxdefiant, 2022-08-01:
recorder.log2.csv is restored from backup. recorder.log4.csv is a new recording that has been created after restoring the backup.

gfwilliams, 2022-08-01:
I mean, when you did `head -n 3 /tmp/test/storage-files/recorder.log2.csv` is that from the backup zip?

So are we saying that the file in the zip file is fine, but when it gets written back, it's actually just a sequence of CSV values on the Bangle?

Because if so I can maybe see how this happens

nxdefiant, 2022-08-01:
ah sorry, yes thats the extracted backup.

Yes, this is what it looks like. Its ends up as comma separated byte values.

gfwilliams, 2022-08-01:
Ok, great. reproduced - you can get it to happen by uploading this file:

[test.zip](https://github.com/espruino/BangleApps/files/9234777/test.zip)

I'll try and get a fix in

gfwilliams, 2022-08-01:
Ok, sorted - turns out it only happened if files had certain contents in (if it thought it was faster to try and send them compressed).

If you restore your backup with the 'development' app loader then you should be fine

closed 2022-08-01 by gfwilliams

gfwilliams, 2022-08-02:
Pretty sure it works with the file I uploaded before?

Maybe you could upload your `storage-files/recorder.log2.csv` (the one with newlines in) so I can test with that?

nxdefiant, 2022-08-02:
I did another test run with recorder.log0.csv as the only file in storage-files, having only the 121 lines. So not leaking any GPS coordinate.

[recorder.log0.csv](https://github.com/espruino/BangleApps/files/9244346/recorder.log0.csv)

This is the result:
[image]

With the longer files I get more of this undefined values, but this should be good to reproduce the problem.

gfwilliams, 2022-08-10:
Please can you try a backup with just this file? [storage-files.zip](https://github.com/espruino/BangleApps/files/9299281/storage-files.zip)

This is the file you gave me, in a zip. I've just done a 'restore' with it a few times, and each time I see the exact file - no `undefined` or anything.

If you still get a problem, please can you go to the Chrome debug console, type `Puck.debug=3`, do a restore and then paste up what it shows?

nxdefiant, 2022-08-10:
ok, after uploading storage-files.zip with only recorder.log0.csv the file is ok. I have tested it three times.

But when uploading with other files I still get the undefined.

nxdefiant, 2022-08-10:
A backup/restore of default apps with recorder0.csv is also ok. So it looks like it is something in my backup.zip (or just its size) that is causing this problem?

nxdefiant, 2022-08-11:
I guess it would be helpful that I send you my backup that fails. Can I send it to the espruino contact@ e-mail?

gfwilliams, 2022-08-11:
Thanks for checking! At least that bit does work now.

yes, getting the full backup would be a good idea. Might help me debug it a bit easier. gw@pur3.co.uk is best

gfwilliams, 2022-08-11:
Thanks! Ok, just narrowed it down to:

```
b=require("heatshrink").decompress(atob("qlpttlllMsN0tN0utkBAVvt1nBItBtgSFhUxm01mc1m05mUymcsAAINImYNMmgNMAoINLmwNMm4NMnANMnINLmcwBpkxBpiLMRQINMRZgDBBpiLMmaLMmaLMmaLMmiLMmiLMmiLMPgINMRZk0RZk0RZk0RZk0RZk0RZk1RZk1RZk1RZgEBBpiLMmqLMAQINMRZk1RZk1RZk2RZk2RZk2RZhgBBpiLMAoQNLRZk2RZk2RZgBBBpc3RZk3RZk3RZhSBBpiLMm6LMm6LMm6LMm6LMm6LMnCLMnCLMnCLMIgINMRZk4RZk4RZk4RZk4RZk4RZk5RZk5RZk5RYQA="))
// b = 1024 element array
E.toString(b);
// returns undefined!
```

But only after all the other upload that has come before.

`E.toString` tries to make a flat string out of the given array, and in this case it can't because it can't find a clear contiguous area of memory (because a 'command history' is stored and it's taking up space).

It's a bit tricky - we don't *need* a flat string, but `E.toString` is the only way to efficiently convert the arraybuffer from heatshrink to one.

Anyway, I've fixed this on the latest cutting edge builds by not storing history, and I'll see about tweaking the app loader so it just doesn't use heatshrink for storagefile uploads (which would solve this problem for everything at the expense of very slightly slower uploads)

nxdefiant, 2022-08-11:
Awesome find, thanks.