espruino/BangleApps #2073: [health] Download CSV doesn't do anything

smatthie, 2022-08-03:
### Affected hardware version

Bangle 2

### Your firmware version

2v14

### The bug

In Gadgetbridge, in the app list, when I click on the "save" icon, it shows the monthly data files on the watch with a choice "download CSV" and "delete".

Clicking on "Download CSV" gives a message "Please wait - downloading health info" but then nothing happens, and I can't find any downloaded CSV on my device. It does not show a directory selection dialogue or any indication where the file was downloaded to. 

Expected behaviour: It should show a dialogue to select the location of the download and then actually create a CSV file there.

I don't know if this is an issue of the health app or the gadgetbridge app as I don't understand which app is actually responsible for this. This is on a Fairphone 4 with /e/OS (Android 11).

gfwilliams, 2022-08-04:
Please can you try downloading via https://banglejs.com/apps on a PC and see if it works there just in case? I just tried here on a PC and it works fine, but I believe I'm seeing this on Gadgetbridge too. 

> Clicking on "Download CSV" gives a message "Please wait - downloading health info" but then nothing happens

So the `Please wait - downloading health info` message disappears after?

I think the issue here might be that the WebView in Gadgetbridge may currently be set up such that it can't display any dialogs to choose files to read/write to.

smatthie, 2022-08-04:
Thanks. Yes, the "Please wait" message dissappears after a while and it's back to the list of months and the buttons "download" and "delete".

I tried it on PC now and it works as expected, so it's really only an issue in the Gadgetbridge app.

gfwilliams, 2022-08-04:
Ok, thanks. Just filed an issue on the Gadgetbridge port with https://github.com/gfwilliams/Gadgetbridge/issues/6

bobrippling, 2023-11-15:
Closing this as the issue is on [gadgetbridge](https://github.com/gfwilliams/Gadgetbridge/issues/6)

closed 2023-11-15 by bobrippling

gfwilliams, 2023-11-16:
Thanks! Looks like I fixed this anyway https://codeberg.org/Freeyourgadget/Gadgetbridge/commit/c647c2647f1b14369edaf5d4597f14c369ec0510