espruino/BangleApps #2075: [run application] question, bug?

pglez82, 2022-08-05:
Hi.
Yesterday I tried the run application in my banglejs1. The application looks very nice and after fixing the position and pressing BTN2, it started recording my running session. The issue is that the interface gets frozen while recording, not updating the time, pace or heartbeat rate. Only when I stop recording by pressing BTN2 again I can see some data about the session.

Is this behaviour expected?

Regards,
Pablo

gfwilliams, 2022-08-09:
> Firmware version: 2v12

Please can you update your firmware and report back? This is extremely likely to be your problem (that the run app relies on some features that were introduced in 2v13/2v14 firmwares)

pglez82, 2022-08-15:
I have updated to 2v15 and the application is running fine. I can close this.
Thanks.

closed 2022-08-15 by pglez82