espruino/BangleApps #2078: [Android][Messages] GadgetBridge

dapgo, 2022-08-08:
### Affected hardware version

Bangle 1

### Your firmware version

2v14

### The bug

Messages 0.49
Android 0.15

BT connection con Gadgetbrige v0.67 is stablished,  BUT notifications or alarms are nor transmitted, warning message on GB "Gadgetbridge plugin not installed on Bangle.js"

After upgrading fdroid GB to v0.68 , warning is still and notifications don't work.

workaround: Settings, Utils, Debug info (change value i.e 'show'), back,back
(Updating boot0)

It seems that the boot0 file  requires to be reloaded

gfwilliams, 2022-08-09:
Interesting, thanks - is it at all reproducible though? What actually started off the behaviour for you?

As you say it sounds like for some reason `.boot0` got deleted/corrupted - I guess the `settings -> utils -> rewrite settings` may have fixed it too?

dapgo, 2022-08-09:
If there is not a common issue, then maybe i know the cause..

I remember that during the installation of "messages" from chrome-android the installation got stuck, (I suffered same issue with other apps time ago; failing from android but installing well from desktop).

So i had to reset the bangle, reload app loader and reinstall it, and the second time the installation succeed but it got the error with notification.

Then removed messages and android, turn off, turn on, reinstall again but same error.
Then i modified settings and it began working

gfwilliams, 2022-08-10:
Yes, I'm pretty sure it's just something that got messed up for you. If Gadgetbridge wasn't working in general there'd be all kinds of  complaints!

I'll close this for now but if anyone else has this in normal use let's open it again.

closed 2022-08-10 by gfwilliams