espruino/BangleApps #2080: [Sleep log] request for compatibility with bangle.js1

dapgo, 2022-08-08:
Currently "Sleep Log" is no available at the App Loader for banglejs1. 
Is there any hardware dependency or technical limitation that will avoid it working  on banglejs1? 

BTW, forcing installation it produces following errors.

Uncaught Error: Unknown font "12x20"
 at line 7 col 682 in sleeplog.app.js
...);g.reset().setFont("12x20").setFontAlign(0,-1);g.drawString...
                              ^
in function "drawNightTo" called from line 8 col 86 in sleeplog.app.js
...true)||{});drawNightTo(prevDays);Bangle.setUI("leftright",(cb)=>{...
                                   ^
in function "startApp" called from line 14 col 10 in sleeplog.app.js
startApp();

gfwilliams, 2022-08-09:
The 12x20 font issue is pretty easily solved, *but* my understanding is that sleep log uses *lots* of RAM. Enough that it can be an issue even on Bangle.js 2.

So I think even if the font issue were resolved, it still wouldn't be usable

dapgo, 2022-11-30:
I just saw the screenshots and the doc about settings and yes definitely it is too much for a bangle.js1.  It is a sophisticated standalone app. :o

However, if in the future sleeplog gets able to sync with GB, then it would be great to just to fork the sleep monitor function so data can be compiled and sent to GB. I understand that banglejs1 can cope with this.