espruino/BangleApps #2086: [agenda] all-day event date is offset by -1 day

m-p-3, 2022-08-12:
### Affected hardware version

Bangle 2

### Your firmware version

2v15

### The bug

### Describe the bug
The Agenda app is displaying all-day events offset by -1 day.

### Expected behavior
Display the correct start date when compared to the event in the Agenda

### Steps to reproduce
Create an all-day event in Google Calendar
Title: `Test (All-day)`
☑️ All-day
Start date: `Sun., Aug. 14th, 2022`
End date: `Sun., aug. 14th, 2022`

**Google Calendar**
[image]

**List of events on device**
[image]

**Details of the event on device**
[image]

gfwilliams, 2022-08-15:
Ok, great - thanks for the report! Any thoughts @glemco ?

As far as I can tell, this might be a tweak we need to make in Gadgetbridge itself

glemco, 2022-08-15:
That is extremely odd.. All day events are handled by gadgetbridge in a peculiar way: they are set as regular event starting at 0:00 of the event's day and ending at 0:00 of the following day (probably because of a pebble's limitation I didn't bother to inspect).
I can confirm that this is the behaviour on my setup, I don't have google calendar so I cannot test that.
My rough guess is that there's a problem with the time zone: are you perhaps at UTC+12 (or whatever it's called)?
Are you sure both the bangle and GB/android have the same (and correct) time zone? I know that should be automatic but it's worth trying

m-p-3, 2022-08-15:
My timezone is set to -4 on the BangleJS2, same on my phone in the OS settings (GMT-04:00 Eastern Daylight Time).

I don't see any settings in Gadgetbridge that lets me define a timezone so far (I assume GB fetches the timezone of the OS), and it has the ability to acquire my location.

glemco, 2022-08-15:
Mmh, that explains at least the 12 (I suppose you use a 12 hours format, not showing am/pm in the agenda makes the things worse, I'll look into that too, you may have realized that agenda is a little too simple).
Anyway I couldn't reproduce the bug, could you please send the ICS file for that event (possibly exported by google calendar) and the bangle's calendar database (`android.calendar.json`)? You can of course remove all non relevant events (it's a simple json), I'm just interested in that test event.
I still think it's something related to the time zone, but reproducing your setup should help.
I might do something smarter to show all-day events but I need to understand which timestamp is actually meaningful.

gfwilliams, 2022-08-16:
In Gadgetbridge on the phone you could also do `Debug -> Fetch device debug logs` (which writes a file to Gadgetbridge's folder on your phone's storage) after you've connected to your Bangle, and then we'd be able to see exactly what Gadgetbridge sent to the watch, so we could figure out for sure what is at fault.

m-p-3, 2022-08-16:
Here are my Bangle.JS2 locale settings
* Time Zone: -4
* Time Format: 24h
* Start Week On: Sun

Here's a sample of the event from Google Calendar, named *Test Event (All-Day)* set on 2022-08-17 (tomorrow)  

I couldn't find a way to manually export a single event to ICS, so I added another email of mine that isn't on Google to get an `event.ics` file, so here's the anonymized content.

```
BEGIN:VCALENDAR
PRODID:-//Google Inc//Google Calendar 70.9054//EN
VERSION:2.0
CALSCALE:GREGORIAN
METHOD:REQUEST
BEGIN:VEVENT
DTSTART;VALUE=DATE:20220817
DTEND;VALUE=DATE:20220818
DTSTAMP:20220816T141221Z
ORGANIZER;CN=**********@gmail.com:mailto:**********@gmail.com
UID:7nmjdocpkmg0pl6oamdsj5frui@google.com
ATTENDEE;CUTYPE=INDIVIDUAL;ROLE=REQ-PARTICIPANT;PARTSTAT=NEEDS-ACTION;RSVP=
 TRUE;CN=**********@protonmail.com;X-NUM-GUESTS=0:mailto:**********@pr
 otonmail.com
ATTENDEE;CUTYPE=INDIVIDUAL;ROLE=REQ-PARTICIPANT;PARTSTAT=ACCEPTED;RSVP=TRUE
 ;CN=**********@gmail.com;X-NUM-GUESTS=0:mailto:**********@gmail.com
X-MICROSOFT-CDO-OWNERAPPTID:-919500491
CREATED:20220816T141049Z
DESCRIPTION:-::~:~::~:~:~:~:~:~:~:~:~:~:~:~:~:~:~:~:~:~:~:~:~:~:~:~:~:~:~:~
 :~:~:~:~:~:~:~:~::~:~::-\nDo not edit this section of the description.\n\nV
 iew your event at https://calendar.google.com/calendar/event?action=VIEW&ei
 d=N25tamRvY3BrbWcwcGw2b2FtZHNqNWZydWkgbWF0aGV3LnBpbmFyZEBwcm90b25tYWlsLmNvb
 Q&tok=MjMjbWF0aGV3LnBpbmFyZEBnbWFpbC5jb21hZjhmNTFhYzExNjNmOTY3ZTkyOGZkOWNhM
 DZiMjI1MjE5Yzc4NzA5&ctz=America%2FToronto&hl=en_GB&es=1.\n-::~:~::~:~:~:~:~
 :~:~:~:~:~:~:~:~:~:~:~:~:~:~:~:~:~:~:~:~:~:~:~:~:~:~:~:~:~:~:~::~:~::-
LAST-MODIFIED:20220816T141221Z
LOCATION:
SEQUENCE:0
STATUS:CONFIRMED
SUMMARY:Test Event (All-Day)
TRANSP:TRANSPARENT
END:VEVENT
END:VCALENDAR

```

Here's the relevant section in `android.calendar.json`

```
{"t":"calendar","id":113,"type":0,"timestamp":1660665600,"durationInSeconds":86400,"title":"Test Event (All-Day)","description":"","location":"","allDay":false}
```

glemco, 2022-08-16:
Mmh, I will make some more tests once I have a little bit of time, but it looks like all-day events from gcalendar and the calendar I use (etar) look quite different, and GB fails identifying the allDay for both (or at least that's what is saved on the bangle). Etar does the conversion directly and uses 0:00 as start and end time, just like GB is supposed to do, probably not having a time (as it happens in gcalendar) is puzzling the system a little.
By the way, if I import this event in my calendar and try to edit, also etar gets it wrong (recognizes it as all-day but freaks out when modifying it, in any case it looks like bangle receives it somehow correct). My rough guess is that either there's no well-defined standard for all-day events or either of the apps isn't exactly compliant.
Anyway, I'll try to inspect this a little more (also on GB side) and see if we can have a workaround. In the worst case, something like assuming all events lasting exactly n days are allday or something like that.

glemco, 2022-08-21:
Found a problem in the way GB is handling the all-day events. Basically it reads it properly from the android store but loses it on the way before uploading to the device. I already sent a PR for that.
In the meanwhile I'm trying to make the agenda aware of that field, perhaps there are also other problems in the way the app is displaying the date with respect to the time zone, I'll release a new version as soon as I'm sure it works also with `UTC-X` (I'm in `UTC+1`, maybe I'm just lucky it works).

gfwilliams, 2022-09-09:
Fixed with this merge: https://codeberg.org/Freeyourgadget/Gadgetbridge/pulls/2836

as per https://github.com/espruino/BangleApps/pull/2117#issuecomment-1240563481 - we just need a new Gadgetbridge build now

closed 2022-09-09 by gfwilliams