espruino/BangleApps #2087: [Recorder] Web-Interface cannot download tracks with broken timestamps

NachtigallersLehrling, 2022-08-14:
### Affected hardware version

Bangle 2

### Your firmware version

2v14.149

### The bug

### Steps to reproduce
0. Set correct time (somewhere in 2022), disconnect from Gadgetbridge
1. Start recording
2. Wait until battery is empty
3. Charge Bangle
4. Boot  (Time is now in 1970 and stays there until connected to Gadgetbridge, dont fix it now)
5. Let recorder record more data (with timestamps in 1970)
6. Stop recording
7. Connect with Apploader
8. Try to download track

### Bug
The track wont download

### Expected behaviour
The timestamps in 1970 should not prevent the whole track from downloading

gfwilliams, 2022-08-15:
Would you be able to upload the recorded file so we can try here please?

Just got to https://www.espruino.com/ide/ and connect (top left) then click the storage icon (4 discs, middle of screen) and scroll down and something like `recorder.log.0.csv (Storagefile)` should be listed. Assuming it's not too big the IDE should be able to download it to your PC and then you can upload here

NachtigallersLehrling, 2022-08-15:
I already deleted the two files affected so far, but i can upload the next affected file

gfwilliams, 2022-08-16:
That would be great, thanks!

I just tried here with something very basic - running recorder, then stopping resetting the time and starting again, and I get a simple CSV file with dates back to 1920 that does download fine - so it's not specifically date related:

```
Time,Latitude,Longitude,Altitude
2022-08-16T08:55:56.000Z,,,
2022-08-16T08:56:06.000Z,,,
1970-01-01T00:00:19.000Z,,,
1970-01-01T00:00:29.000Z,,,
```

NachtigallersLehrling, 2022-08-16:
Huh, strange. I tried to download another track today in which I let the battery drain by purpose but this time I could download it just fine. I assume the problem isn't quite what I thought, sorrry about that. Thanks for checking, I will come back to this once it happens again.

bobrippling, 2023-11-15:
@NachtigallersLehrling I'll close this for the meanwhile but please reopen if you find it happening again