espruino/BangleApps #2101: Widges not working anymore

kmg454, 2022-08-25:
Upgraded my watch with the latest firmware and lost all my apps and widgets.
So installed all the default apps and the watch was working again.
But when tried to install my widgets I got problems.
Non of the widgets are working anymore because the error say

Uncaught ReferenceError: "WIDGETS" is not defined
 at line 76 col 16 in .bootcde
  WIDGETS["hrm"]={area:"tl",width:65,draw:draw};

thyttan, 2022-08-26:
[Related to this espruino forum post](http://forum.espruino.com/conversations/379195/).

gfwilliams, 2022-09-05:
Sorry for the delay... I just replied in that post, but I'm pretty sure that:

> But when tried to install my widgets I got problems.

It seems that you may have copied it in using the Web IDE while it was set to upload to Flash (rather than a file in Bangle.js storage), which overwrote the bootloader?

The Bangle.js development docs (http://www.espruino.com/Bangle.js+Develo­pment) mention:

> Note: If you've used Espruino before, make sure the Upload button (middle of the screen) shows RAM underneath it. Otherwise by writing to Flash you can remove the watch's built-in menu (you can just re-add it by removing and installing bootloader using [https://banglejs.com/apps](http://microco.sm/out/Yb4Yh))

Please can you confirm if that fixes it so I can close this issue if so?

kmg454, 2022-09-05:
I tried uploading to ram or flash but same problem message Uncaught ReferenceError: "WIDGETS" is not defined
at line 76 col 16 in .bootcde
WIDGETS["hrm"]={area:"tl",width:65,draw:draw};

gfwilliams, 2022-09-05:
> I tried uploading to ram or flash

Yes, but **once you have uploaded to flash, you have to reinstall the bootloader to fix it**. Just changing to upload to RAM will then have no effect.

kmg454, 2022-09-05:
I uploaded and after reinstall the bootloader the widget is gone

kmg454, 2022-09-05:
No it is not working at all, I want upload my widgets on a normal way like I did before, none of the widgets I used before are working. When I upload to flash or Ram I know the widgets are working because I see the data on my BLE hub coming in. But the watch is not booting just showing the logo and a message say Bluetooth. After repair the boot loader the watch is booting again but the widgets are gone

gfwilliams, 2022-09-05:
> After repair the boot loader the watch is booting again but the widgets are gone

Yes, this is what you'd expect

> No it is not working at all

And you have definitely uploaded your widget to a file called `mywidget.wid.js` ? 

Some apps don't load Widgets, so that could be what you're hitting?

kmg454, 2022-09-05:
Tried a second time and seems to work again. Thanks