espruino/BangleApps #2124: [agpsdata] Debug info "error: No http method found"

thyttan, 2022-09-15:
### Affected hardware version

Bangle 2

### Your firmware version

2v15.14 (Also had it on 2v15)

### The bug

Description:
When A-GPS Data is installed and the Bangle is connected to Web IDE the message "error: No http method found" is often printed.

Expected behavior:
To there not to be an error.

Steps to reproduce:
1. Connect Bangle to Web IDE
2. Install [A-GPS Data](https://banglejs.com/apps/?q=a-gps%20data)
3. "error: No http method found" is printed in the Web IDE.

The message is often displayed when launching different apps as well.

myxor, 2022-09-16:
This is triggered in https://github.com/espruino/BangleApps/blob/674ef0aa133a22724c532b11e8c51445ed411c01/apps/agpsdata/lib.js#L72 because the WebIDE does not (yet*) support the HTTP call functionality.

Which confuses me is that you say you see this log message often. This should not be.

*: there is somewhere the idea to extend the WebIDE with this functionality.

gfwilliams, 2022-09-16:
Do you have the `Android Integration` app installed as well? You need that for the A-GPS Data app to work

thyttan, 2022-09-17:
@myxor 
If I am connected to the Web IDE on my laptop, I seem to get the error every time I load any app.

@gfwilliams 
Yes, Android Integration installed. When I'm connected to gadgetbridge everything seems to work as expected 👍.

Feel free to close this if we're done! :)

thyttan, 2022-09-17:
A little OT:
@myxor @pidajo 
Should "A-GPS Data" be added as a dependency for "AGPS Widget"? And maybe the descriptions could be clearer about if they are dependent on each other or not?

myxor, 2022-09-18:
> A little OT: @myxor @pidajo Should "A-GPS Data" be added as a dependency for "AGPS Widget"? And maybe the descriptions could be clearer about if they are dependent on each other or not?

They are independent from each other.

gfwilliams, 2022-09-21:
> because the WebIDE does not (yet*) support the HTTP call functionality.

The Android Integration app is loaded regardless of if you're in the IDE or not, so `Bangle.http` should exist and you shouldn't get the error.

Maybe when you're in the IDE you could check if `Bangle.http` exists?

>  And maybe the descriptions could be clearer about if they are dependent on each other or not?

Ideally, yes. I'll add something - because honestly I feel like 99% of people just want `Assisted GPS Update (AGPS)` but the other apps confuse things a bit

thyttan, 2022-09-21:
>Maybe when you're in the IDE you could check if Bangle.http exists?

Doing
```
>Bangle.http
```
in the Web Ide returns
```
=function (url,options) { ... }
```
.

gfwilliams, 2022-09-21:
Ok, thanks. Ok, well I think I understand now then!

@myxor I'm pretty sure the issue is https://github.com/espruino/BangleApps/blob/674ef0aa133a22724c532b11e8c51445ed411c01/apps/agpsdata/boot.js#L23 - the code is going to be running as 'boot code' - right at the very start before other things have initialised.

In some cases the install order will be different and the Android Integration won't have even have run yet, so `Bangle.pull` won't exist.

The error will be happening all the time (whether connected to Gadgetbridge or not) but because there is a `setInterval` there, the code that runs later in `setInterval` will work fine.

I think there are two options:

* wrap that code in a `setTimeout(..., 10)` so that it runs after all the rest of the init code, and you can be sure `Bangle.pull` is there.
* Just delete the code, and rely on `setInterval`. This'll make apps boot faster, and realistically if you're checking every few minutes, that's going to be more than good enough.

I should add though that since the AGPS data only updates once a day, I'm not sure giving folks the option to query the Espruino server every minute is really in anyone's best interests? It'll increase bandwidth usage on your phone and slow down the Bangle, not to mention hammering my server if everyone installs it.

thyttan, 2022-09-27:
http://forum.espruino.com/conversations/380123/#16701648

HughB's espruino forum post comment above seems to indicate AGPS data 'should' be updated every six hours?

gfwilliams, 2022-09-28:
This is fixed with https://github.com/espruino/BangleApps/pull/2149 now - thanks @myxor!

closed 2022-09-28 by gfwilliams