espruino/BangleApps #2128: [Rebel] S0.09 fails to load, error on start up

hughbarney, 2022-09-18:
### Affected hardware version

Bangle 2

### Your firmware version

2v15

### The bug

Rebel fails to load, just hangs
Get error on IDE:

Uncaught Error: Cannot read property 'autoCycle' of undefined
 at line 48 col 11 in rebble.app.js
if(settings.autoCycle||settings.sideTap==0){Bangle.setUI("clock...
             ^

Rarder44, 2022-09-18:
I don't know why but there was a push linked to an old issue 12 days ago, which moved some pieces of code (no insertion / removal)

thyttan, 2022-09-19:
#2103: Background on why code was moved around so setUI is called before loadWidgets. It's so that the widgets will know if the active app is a clock or not, and maybe other things.
#2113: This is the PR that moved the code around for the rest of the clock apps.

gfwilliams, 2022-09-20:
Fixed with https://github.com/espruino/BangleApps/pull/2130 - thanks for the speedy fix

closed 2022-10-12 by gfwilliams