espruino/BangleApps #2152: [owmweather] Crazy battery consumption if not connected

glemco, 2022-09-27:
### Affected hardware version

Bangle 2

### Your firmware version

2v15

### The bug

The bangle's battery is drained in about a day without being connected to the phone, also it seems like the connection becomes unstable although the phone is reachable.

I set the owmweather app to fetch data every 3h and it seems like it works fine, however it appears to me that it tries to fetch data from the device also if not connected. By quickly checking the code, everything looks normal, however when I connect the watch back to the phone, I see the logcat cluttered with messages of this type:
```
09-27 09:50:58.480 16424 16446 D nodomain.freeyourgadget.gadgetbridge.service.btle.BtLEQueue: characteristic write: 6e400002-b5a3-f393-e0a9-e50e24dcca9e (success)
09-27 09:50:58.481 16424  6037 D nodomain.freeyourgadget.gadgetbridge.service.btle.BtLEQueue: About to run action: 27 settembre, 09:50: WriteAction on characteristic: 6e400002-b5a3-f393-e0a9-e50e24dcca9e
09-27 09:50:58.482 16424  6037 D nodomain.freeyourgadget.gadgetbridge.service.btle.actions.WriteAction: writing to characteristic: 6e400002-b5a3-f393-e0a9-e50e24dcca9e: 0x31 0x34 0x39 0x33 0x30 0x31 0x35 0x35 0x34 0x22 0x2c 0x65 0x72 0x72 0x3a 0x22 0x49 0x6e 0x74 0x65
09-27 09:50:58.484 16424 16446 D nodomain.freeyourgadget.gadgetbridge.service.btle.BtLEQueue: characteristic write: 6e400002-b5a3-f393-e0a9-e50e24dcca9e (success)
09-27 09:50:58.485 16424  6037 D nodomain.freeyourgadget.gadgetbridge.service.btle.BtLEQueue: About to run action: 27 settembre, 09:50: WriteAction on characteristic: 6e400002-b5a3-f393-e0a9-e50e24dcca9e
09-27 09:50:58.486 16424  6037 D nodomain.freeyourgadget.gadgetbridge.service.btle.actions.WriteAction: writing to characteristic: 6e400002-b5a3-f393-e0a9-e50e24dcca9e: 0x72 0x6e 0x65 0x74 0x20 0x61 0x63 0x63 0x65 0x73 0x73 0x20 0x6e 0x6f 0x74 0x20 0x65 0x6e 0x61 0x62
09-27 09:50:58.488 16424 16446 D nodomain.freeyourgadget.gadgetbridge.service.btle.BtLEQueue: characteristic write: 6e400002-b5a3-f393-e0a9-e50e24dcca9e (success)
09-27 09:50:58.489 16424  6037 D nodomain.freeyourgadget.gadgetbridge.service.btle.BtLEQueue: About to run action: 27 settembre, 09:50: WriteAction on characteristic: 6e400002-b5a3-f393-e0a9-e50e24dcca9e
09-27 09:50:58.490 16424  6037 D nodomain.freeyourgadget.gadgetbridge.service.btle.actions.WriteAction: writing to characteristic: 6e400002-b5a3-f393-e0a9-e50e24dcca9e: 0x6c 0x65 0x64 0x2c 0x20 0x63 0x68 0x65 0x63 0x6b 0x20 0x47 0x61 0x64 0x67 0x65 0x74 0x62 0x72 0x69
09-27 09:50:58.493 16424 16446 D nodomain.freeyourgadget.gadgetbridge.service.btle.BtLEQueue: characteristic write: 6e400002-b5a3-f393-e0a9-e50e24dcca9e (success)
09-27 09:50:58.493 16424  6037 D nodomain.freeyourgadget.gadgetbridge.service.btle.BtLEQueue: About to run action: 27 settembre, 09:50: WriteAction on characteristic: 6e400002-b5a3-f393-e0a9-e50e24dcca9e
09-27 09:50:58.495 16424  6037 D nodomain.freeyourgadget.gadgetbridge.service.btle.actions.WriteAction: writing to characteristic: 6e400002-b5a3-f393-e0a9-e50e24dcca9e: 0x64 0x67 0x65 0x20 0x44 0x65 0x76 0x69 0x63 0x65 0x20 0x53 0x65 0x74 0x74 0x69 0x6e 0x67 0x73 0x22
09-27 09:50:58.497 16424 16446 D nodomain.freeyourgadget.gadgetbridge.service.btle.BtLEQueue: characteristic write: 6e400002-b5a3-f393-e0a9-e50e24dcca9e (success)
09-27 09:50:58.497 16424  6037 D nodomain.freeyourgadget.gadgetbridge.service.btle.BtLEQueue: About to run action: 27 settembre, 09:50: WriteAction on characteristic: 6e400002-b5a3-f393-e0a9-e50e24dcca9e
09-27 09:50:58.498 16424  6037 D nodomain.freeyourgadget.gadgetbridge.service.btle.actions.WriteAction: writing to characteristic: 6e400002-b5a3-f393-e0a9-e50e24dcca9e: 0x7d 0x29 0x0a
09-27 09:50:58.500 16424 16446 D nodomain.freeyourgadget.gadgetbridge.service.btle.BtLEQueue: characteristic write: 6e400002-b5a3-f393-e0a9-e50e24dcca9e (success)
09-27 09:50:58.577 16424 16446 D nodomain.freeyourgadget.gadgetbridge.service.btle.BtLEQueue: characteristic changed: 6e400003-b5a3-f393-e0a9-e50e24dcca9e value: 0x0d 0x0a 0x7b 0x22 0x74 0x22 0x3a 0x22 0x68 0x74 0x74 0x70 0x22 0x2c 0x22 0x75 0x72 0x6c 0x22 0x3a
09-27 09:50:58.577 16424 16446 I nodomain.freeyourgadget.gadgetbridge.service.devices.banglejs.BangleJSDeviceSupport: RX: 
09-27 09:50:58.577 16424 16446 I nodomain.freeyourgadget.gadgetbridge.service.devices.banglejs.BangleJSDeviceSupport: {"t":"http","url":
09-27 09:50:58.578 16424 16446 I nodomain.freeyourgadget.gadgetbridge.service.devices.banglejs.BangleJSDeviceSupport: UART RX LINE: 
09-27 09:50:58.587 16424 16446 D nodomain.freeyourgadget.gadgetbridge.service.btle.BtLEQueue: characteristic changed: 6e400003-b5a3-f393-e0a9-e50e24dcca9e value: 0x22 0x68 0x74 0x74 0x70 0x73 0x3a 0x2f 0x2f 0x61 0x70 0x69 0x2e 0x6f 0x70 0x65 0x6e 0x77 0x65 0x61
09-27 09:50:58.587 16424 16446 I nodomain.freeyourgadget.gadgetbridge.service.devices.banglejs.BangleJSDeviceSupport: RX: "https://api.openwea
09-27 09:50:58.598 16424 16446 D nodomain.freeyourgadget.gadgetbridge.service.btle.BtLEQueue: characteristic changed: 6e400003-b5a3-f393-e0a9-e50e24dcca9e value: 0x74 0x68 0x65 0x72 0x6d 0x61 0x70 0x2e 0x6f 0x72 0x67 0x2f 0x64 0x61 0x74 0x61 0x2f 0x32 0x2e 0x35
09-27 09:50:58.598 16424 16446 I nodomain.freeyourgadget.gadgetbridge.service.devices.banglejs.BangleJSDeviceSupport: RX: thermap.org/data/2.5
09-27 09:50:58.600 16424 16445 D nodomain.freeyourgadget.gadgetbridge.service.btle.BtLEQueue: characteristic changed: 6e400003-b5a3-f393-e0a9-e50e24dcca9e value: 0x2f 0x77 0x65 0x61 0x74 0x68 0x65 0x72 0x3f 0x6c 0x61 0x74 0x3d 0x34 0x39 0x2e 0x34 0x33 0x26 0x6c
09-27 09:50:58.601 16424 16445 I nodomain.freeyourgadget.gadgetbridge.service.devices.banglejs.BangleJSDeviceSupport: RX: /weather?lat=LLLLL&l
09-27 09:50:58.609 16424 16446 D nodomain.freeyourgadget.gadgetbridge.service.btle.BtLEQueue: characteristic changed: 6e400003-b5a3-f393-e0a9-e50e24dcca9e value: 0x6f 0x6e 0x3d 0x37 0x2e 0x37 0x35 0x26 0x65 0x78 0x63 0x6c 0x75 0x64 0x65 0x3d 0x68 0x6f 0x75 0x72
09-27 09:50:58.609 16424 16446 I nodomain.freeyourgadget.gadgetbridge.service.devices.banglejs.BangleJSDeviceSupport: RX: on=LLLL&exclude=hour
09-27 09:50:58.613 16424 16445 D nodomain.freeyourgadget.gadgetbridge.service.btle.BtLEQueue: characteristic changed: 6e400003-b5a3-f393-e0a9-e50e24dcca9e value: 0x6c 0x79 0x2c 0x64 0x61 0x69 0x6c 0x79 0x26 0x61 0x70 0x70 0x69 0x64 0x3d 0x35 0x32 0x66 0x31 0x64
09-27 09:50:58.613 16424 16445 I nodomain.freeyourgadget.gadgetbridge.service.devices.banglejs.BangleJSDeviceSupport: RX: ly,daily&appid=XXXX
09-27 09:50:58.623 16424 16445 D nodomain.freeyourgadget.gadgetbridge.service.btle.BtLEQueue: characteristic changed: 6e400003-b5a3-f393-e0a9-e50e24dcca9e value: 0x64 0x65 0x63 0x39 0x34 0x33 0x33 0x33 0x62 0x30 0x62 0x33 0x61 0x61 0x30 0x35 0x38 0x37 0x30 0x31
09-27 09:50:58.623 16424 16445 I nodomain.freeyourgadget.gadgetbridge.service.devices.banglejs.BangleJSDeviceSupport: RX: XXXXXX
09-27 09:50:58.629 16424 16445 D nodomain.freeyourgadget.gadgetbridge.service.btle.BtLEQueue: characteristic changed: 6e400003-b5a3-f393-e0a9-e50e24dcca9e value: 0x61 0x66 0x30 0x64 0x61 0x35 0x37 0x22 0x2c 0x22 0x69 0x64 0x22 0x3a 0x22 0x30 0x32 0x30 0x31 0x32
09-27 09:50:58.629 16424 16445 I nodomain.freeyourgadget.gadgetbridge.service.devices.banglejs.BangleJSDeviceSupport: RX: XXXXX","id":"YYYYYY
09-27 09:50:58.631 16424 16445 D nodomain.freeyourgadget.gadgetbridge.service.btle.BtLEQueue: characteristic changed: 6e400003-b5a3-f393-e0a9-e50e24dcca9e value: 0x32 0x32 0x35 0x39 0x33 0x39 0x22 0x7d 0x0d 0x0a
09-27 09:50:58.631 16424 16445 I nodomain.freeyourgadget.gadgetbridge.service.devices.banglejs.BangleJSDeviceSupport: RX: YYYYY"}
09-27 09:50:58.632 16424 16445 I nodomain.freeyourgadget.gadgetbridge.service.devices.banglejs.BangleJSDeviceSupport: UART RX LINE: {"t":"http","url":"https://api.openweathermap.org/data/2.5/weather?lat=LLLL&lon=LLLL&exclude=hourly,daily&appid=XXXX","id":"YYYYYYYY"}
09-27 09:50:58.632 16424 16445 I nodomain.freeyourgadget.gadgetbridge.service.devices.banglejs.BangleJSDeviceSupport: UART RX JSON parsed successfully
09-27 09:50:58.633 16424 16445 I nodomain.freeyourgadget.gadgetbridge.service.devices.banglejs.BangleJSDeviceSupport: UART TX: GB({t:"http",id:"YYYYYYYYY",err:"Internet access not enabled, check Gadgetbridge Device Settings"})
09-27 09:50:58.634 16424 16445 D nodomain.freeyourgadget.gadgetbridge.service.btle.BtLEQueue: about to add: 09:50:58: Transaction task: http with 6 actions
[... cut ...]
09-27 09:50:58.658 16424 16445 D nodomain.freeyourgadget.gadgetbridge.service.btle.BtLEQueue: characteristic write: 6e400002-b5a3-f393-e0a9-e50e24dcca9e (success)
09-27 09:50:58.734 16424 16445 D nodomain.freeyourgadget.gadgetbridge.service.btle.BtLEQueue: characteristic changed: 6e400003-b5a3-f393-e0a9-e50e24dcca9e value: 0x0d 0x0a 0x7b 0x22 0x74 0x22 0x3a 0x22 0x68 0x74 0x74 0x70 0x22 0x2c 0x22 0x75 0x72 0x6c 0x22 0x3a
```
Did that happen to anyone else? I'm aware I need to test a little bit more to give useful information, anyway I'd avoid flattening again my battery that quick. Let me know what other info I should provide.

myxor, 2022-09-28:
Ping @halemmerich

gfwilliams, 2022-09-28:
Ahh, interesting - thanks! Maybe actually `Bangle.http` should immediately fail if there is no internet connection? I bet what happens is requets get queued up and then maybe eventally the Bangle just ends up stalled (and using power) waiting for the HTTP requests to be sent

gfwilliams, 2022-09-28:
Should be fixed now I hope (if you update the Android app from the dev app loader)

closed 2022-09-28 by gfwilliams

glemco, 2022-09-28:
> Ahh, interesting - thanks! Maybe actually `Bangle.http` should immediately fail if there is no internet connection? I bet what happens is requets get queued up and then maybe eventally the Bangle just ends up stalled (and using power) waiting for the HTTP requests to be sent

Yeah, that was my guess. Sure that's the most elegant solution. Thanks, I'll test it soon!

commit 1b20bb6a (2022-09-28): 0.16: Bangle.http now fails immediately if there is no Bluetooth connection (fix #2152)