espruino/BangleApps #2167: [App Loader] Always installs Dragboard on textinput dependency

thyttan, 2022-10-05:
### Affected hardware version

Bangle 2

### Your firmware version

2v15.29 (had this on previous fw versions as well)

### The bug

Every time I install an app with a dependency on the type 'textinput' the keyboard 'Dragboard' is installed as well, regardless of if another keyboard is already installed or not. The exception is if Dragboard itself is already installed on the Bangle.

If any other keyboard with type 'textinput' specified in its metadata is already installed on the watch Dragboard should not be installed when 'textinput'-type is a dependency.

To reproduce:
1. Install Multitap Keyboard.
2. Install an app with 'textinput'-type dependency, e.g. 'Noteify'.

thyttan, 2022-10-09:
One solution to this could be to add ```"noOverwrite":true``` to the lib.js files entry in metadata.json.  I think this would mean an installed keyboard would not be overwritten in the described scenario.

Keyboards have also been discussed in issue #1682.

gfwilliams, 2022-10-10:
Thanks - I'll look into this. There is code to handle this at:

https://github.com/espruino/EspruinoAppLoaderCore/blob/6857957f5aedfd9d175ecbf8e49d08bb167b8128/js/index.js#L651

So either it's broken or being called before existing apps get loaded

closed 2022-10-10 by gfwilliams