espruino/BangleApps #2168: [E.showMenu] Rendering/UI unreliable when widgets are absent

thyttan, 2022-10-06:
### Affected hardware version

Bangle 2

### Your firmware version

2v15.29 (seen this on earlier fw versions as well)

### The bug

When using menus via E.showMenu with a 'back'-function specified and without having called loadWidgets() it seems confused about how to render title and back button at times, and sometimes when I try to press the first menu item it will not be recognized.

I have the "Swipe menus"-app installed now, but I have experienced similar behavior also with just the regular menus.

Reproduce (is not guaranteed to reproduce the bug, it's hit and miss...):
1. Install and launch [Spotify Remote](https://thyttan.github.io/BangleApps/?q=spotify%20remote).
2. Click 'Srch' in upper right corner.
3. Click 'Search term w/o tags"
4. Exit keyboard by clicking upper left corner or pressing the physical button.
5. Click 'Search term w/o tags" (This step has been the most unreliable for me now, it doesn't always behave badly but quite often.

I don't have a super reliable way to reproduce the bug right now.

gfwilliams, 2022-10-07:
Interesting - thanks! Perhaps it requests the appRect *before* setUI is called, in which case the widget wouldn't be included and it would assume there were no widgets...

thyttan, 2022-10-07:
I've written some test code, my findings are presented as part of the follow along tests a-d:

```
/*
Test code for trying menu system with/without widget field present.

Use Web IDE and upload to ram.

Note the output of Bangle.appRect in the console window.

Test a:
1. Upload to ram.
2. Note that the title "Main menu" flashes quickly before being overwritten by the red back button.
3. Click "1 Submenu".
4. Note that the title "Submenu" is displayed below the red back button.
5. Scroll down to hide the title field completely and then scroll back to the top.
6. Scroll down again.
7. Click the red back button.
8. Note that the title "Main menu" is now displayed below the red back button.
9. Scroll down to hide the title field and then scroll back up.
10. Click the red back button to call E.showMenu(), removing the menu.
11. End test a.

Test b:
1. Upload to ram.
2. Scroll down to hide red back button slightly or completely.
3. Scroll back to the top.
4. Note that the red back button is partially or completely hidden and the title "Main menu" has appeard instead.
5. Click top left corner of the screen where the red back button was before. Note that this still works to execute the back-function.
6. End test b.

Test c:
1. Upload to ram.
2. Scroll down so that "1 Submenu"-entry is slightly hidden.
3. Click "1 Submenu".
4. Note that there is no red back button on top but instead part of the last menu graphic "1 Submenu".
5. Click top left corner of the screen where the red back button was before.
6. End test c.

Test d:
1. Uncomment //Bangle.loadWidgets() down below.
2. Upload to ram.
3. Navigate the menus trying the things in test a-c. Note that they work without a problem in this case.
4. End test d.

*/

print("\n\nBefore code is run:");
print(Bangle.appRect);

var mainMenu = {
  "" : {title : "Main menu",
        back : ()=>{
          print("\n\nBefore calling E.showMenu():");
          print(Bangle.appRect);
          E.showMenu();
          print("After calling E.showMenu():");
          print(Bangle.appRect);
        }
       },
  "1 Submenu" : ()=>{
    print("\n\nBefore calling E.showMenu(subMenu):");
    print(Bangle.appRect);
    E.showMenu(subMenu);
    print("After calling E.showMenu(subMenu):");
    print(Bangle.appRect);
  },
  "2 appRect" : ()=>{
    print("\nOn main menu:");
    print(Bangle.appRect);
  },
  "3" : ()=>{},
  "4" : ()=>{},
  "5" : ()=>{},
  "6" : ()=>{},
  "7" : ()=>{},
  "8" : ()=>{},
};

var subMenu = {
  "" : {title : "Submenu",
        back : ()=>{
          print("\n\nBefore calling E.showMenu(mainMenu):");
          print(Bangle.appRect);
          E.showMenu(mainMenu);
          print("After calling E.showMenu(mainMenu):");
          print(Bangle.appRect);
        }
       },
  "1" : ()=>{},
  "2 appRect" : ()=>{
    print("\n\nOn submenu:");
    print(Bangle.appRect);
  },
  "3" : ()=>{},
  "4" : ()=>{},
  "5" : ()=>{},
  "6" : ()=>{},
  "7" : ()=>{},
  "8" : ()=>{},
};

//Bangle.loadWidgets();

E.showMenu(mainMenu);
```

gfwilliams, 2022-12-05:
Thanks! Sorry for not following this up properly before. Now fixed with [2d3c34ef7c2b9fe2118e816aacd2e096adb99596](https://github.com/espruino/Espruino/commit/2d3c34ef7c2b9fe2118e816aacd2e096adb99596)

closed 2022-12-05 by gfwilliams

thyttan, 2022-12-05:
No problem - thanks!