espruino/BangleApps #2172: [App Loader] Feature suggestion: Add app id's to apps tags automatically

thyttan, 2022-10-06:
Sometimes I will try to search for the app id instead of its name on the App Loader which often returns no result. It would be neat if searching for an app id would show the corresponding app entry on the App Loader.

gfwilliams, 2022-10-07:
Yes, that's a good idea. It always used to do this, and at some point search got tweaked and it stopped

gfwilliams, 2022-10-12:
Ok, should be fixed with https://github.com/espruino/EspruinoAppLoaderCore/compare/be7115722d8e85d1a890585d2812d5ff6079762c...371fa63ae0cd69d5be52a4189400190e86491651#diff-3f689a613b62130589affae189bdfbcff8533286dcf50c288ccf61bd9b59c11c 

search should be WAY better now

closed 2022-10-12 by gfwilliams

thyttan, 2022-10-12:
Very nice, thanks!