espruino/BangleApps #2181: [Settings v0.50 (I think)] New Calibration feature - UI jumps around, can be hard to hit elements, freezes and subsequent reboots

thyttan, 2022-10-19:
### Affected hardware version

Bangle 2

### Your firmware version

2v15.44

### The bug

>When rotation=2 and using "Light switch widget" to change brightness level it works for the most part. But if draging to the very top of the touch screen (higher than the LCD) the brightness will be set to minimum value instead of maximum value - it seems some touch-pixels at the very bottom hasn't followed along in the rotation to become top-pixels, if that makes sense. (From https://github.com/espruino/BangleApps/issues/1040#issuecomment-1276521006)

I get something similar to this on a factory reset watch, updated to fw 2v15.44 doing:
1. I update installed apps. (e.g. Settings v0.50)
2. I try the new 'Calibrate' functionality in settings.
3. When swiping my finger bottom-to-top of screen, the scroller will first scroll down as expected but when reaching the top of the touch screen with my finger the scroller will jump back up and overshoot in the other direction, so I will see empty space above the first menu entry.

The bugs are not always as pronounced, I think it depends on how I press the cross-hairs during calibration. If I aim with my right index finger during calibration I seem to get the bugs more often than if I use my left thumb for example. I think maybe the problems are worse the less "reasonable" my inputs during calibration was.

The red back button can also get much harder to hit reliably. I also get spontaneous freezes and reboots.

(I'll add some video clips)

johan-m-o, 2022-10-20:
I can add my observations on the new Calibrate feature. Running app v0.50 and firmware 2v15.44.

I see the same scrolling behaviour as described here, but I also experience another issue: the inputs have somewhat flipped themselves. Pressing top left registers on bottom right, bottom left registers on top right and the right side of the screen doesn't register any input at all.

gfwilliams, 2022-10-20:
Thanks for the report! Please can you try the absolute latest cutting edge build?

I think what happened is the coordinates were stored in unsigned values, so when you got negative coordinates they just wrapped over to ~255 - so that'd make the scrolling screw up as you've seen, and would also make it very hard to hit the back button. It might also explain why setRotation(2) seemed so broken.

It should be sorted now I hope!

closed 2022-10-20 by gfwilliams

johan-m-o, 2022-10-20:
Confirmed fixed.

gfwilliams, 2022-10-20:
Great, thanks!

@thyttan please could you confirm about https://github.com/espruino/BangleApps/issues/1040#issuecomment-1276521006 too - I think this should fix that!

thyttan, 2022-10-20:
Fixed for me as well. Will come back re/on #1040. 

Just a small note: I had "Light Switch Widget" installed when trying calibrating this time and when I touched the crosshair when in the top right corner, Light switch widget would be displayed and toggle background illumination, instead of registering touch to the crosshair. So I had to try again by touching a little further down on the screen - which worked and the crosshair then moved to the next corner. Should I open a separate issue for this, @gfwilliams?

gfwilliams, 2022-10-20:
Yes, might as well open a new issue. That's a bit frustrating - I guess we can try and remove/rename the old listeners, but I guess another option would be to make the calibration a new app.