espruino/BangleApps #2189: [messages] Missing icons

rigrig, 2022-10-22:
I've kept a list of apps that sent a notification but don't have an icon yet.
I plan to (eventually) submit a PR to add them, but adding them one-by-one felt like way too much churn.

- [ ] KDE Connect
- [ ] Aurora Store
- [ ] Rabobank
- [ ] booq Personeelsplanner
- [ ] NS
- [ ] Network manager
- [ ] f-droid build status
- [ ] KeePassDX
- [ ] phonetrack
- [ ] Mull
- [ ] 1Password
- [ ] Tasks
- [ ] PostNL
- [ ] VLC

@gfwilliams How do you feel about adding lots of apps? Should we just "add them all", or maybe be a bit selective about which apps to include? (I doubt there are many people using "booq Personeelsplanner" or "Rabobank" outside the Netherlands...)

Rarder44, 2022-10-23:
since the list of apps could become long, why not create an online repository with images and the app downloads the icons it needs via a network connection and keeps them in cache?
this would avoid having to commit exclusively to images and the app would be much lighter

gfwilliams, 2022-10-26:
Personally I'd just include all the icons for now - we shouldn't be calling the function that often (and it sits in flash) so I think we'll be fine. If it gets really bad we can always look at including it in binary form.

The problem with the network connection is folks using Apple phones don't have any access to it.

DZPM, 2022-10-26:
How about having broad categories of icons? In example, all the banks could share the same icon, could even be the same bank emoji: 🏦 
There could be categories for password managers, games, etc.

I'm thinking in maintainability. Maintaining this categories would be way easier, as users would only insert "LocalBank" to the list of banks, without having to generate/convert a custom icon each time, or having to look at repeated icons.

gfwilliams, 2022-10-26:
> How about having broad categories of icons?

That sounds like a great idea to me... Given most icons are black and white 24x24px there should really be a lot of overlap.

Maybe one option to make this more maintainable is to have a folder of PNG images named by the app name, and then a simple script that loads them (handling duplicate images) and creates the code from them. It makes it easier to check icons at a glance, but also if at some point we wanted to go for a binary format changing over would be easy too.

m-p-3, 2022-11-17:
Sorry to drop in, but is there some documentation and/or guidelines on the icon format? I might add a couple if possible.

Thanks!

gfwilliams, 2022-11-17:
The icons have now moved to `messageicons`: https://github.com/espruino/BangleApps/blob/master/apps/messageicons/lib.js#L1

But it just says:

> icons should be 24x24px or less with 1bpp colors and 'Transparency to Color'
> http://www.espruino.com/Image+Converter

Hopefully that's enough! If you do add icons, maybe you could add the png of the original icon in a folder? At some point we might change the way we package the icons and then it'd be really handy :)

m-p-3, 2022-11-17:
Thanks, that's helpful :)

> If you do add icons, maybe you could add the png of the original icon in a folder? 

Definitely, good idea!

Rarder44, 2022-11-17:
> The icons have now moved to messageicons

I looked through the code and found that on line 53 the library refers to another app's file ( "messages.settings.json"); is it intentional or should you move the "iconColorMode" setting in the global settings?
If for a module to work fully you need to have another app as well, then maybe it's not better to integrate the library directly into that app? (and don't create a separate app)

gfwilliams, 2022-11-18:
> the library refers to another app's file ( "messages.settings.json")

Yes, it's intentional. If the settings file is not there it'll still work fine, you just can't choose that you want mono icons.

We could add a settings page just for icons, but right now I felt like this was a good compromise.

It's also worth noting that adding a new settings file would mean another file access every time a message icon color was requested, which would be slow. The way it's implemented most places that use the messageicons have already loaded message settings, so that extra read can be avoided.

m-p-3, 2022-11-21:
> If you do add icons, maybe you could add the png of the original icon in a folder?

@gfwilliams I'm almost done tweaking a bunch of icons (93 total in fact), is there a specific path you want for the originals? I can go with `/apps/messageicons/icons_source` of something else if you have a better or less confusing directory name to suggest.

They're all 24x24px PNGs, black-only icons, with transparent background but no alpha-shading, so they should look crisp and should be easy to convert with https://www.espruino.com/Image+Converter as-is.

There's also a couple of generic ones I made, so if storage could be an issue, then maybe we could point some apps to these generic ones. Here's a couple as a sample:

[image] [image] [image] [image] [image] [image] [image] [image]

If those looks good to you, I can upload the rest in the repo.

I'll look into adding them to [lib.js](https://github.com/espruino/BangleApps/blob/master/apps/messageicons/lib.js) once that's done.

Now I'm not sure how to add new entries, as I'm not exactly should how the string variable is populated. I suppose it looks at the notification itself and tries to find a match with the app name (and doesn't look for a Android package name, as that would obviously not work for iOS users)?

Thanks!

gfwilliams, 2022-11-21:
Great! I reckon just `/apps/messageicons/icons` would be good.

>  maybe we could point some apps to these generic ones

Those look great - thanks!

> tries to find a match with the app name

That's right, yes... It's just the user-visible name rather than package name.

Having the icons there as PNGs is great, but I'm a bit iffy about adding 95 icons 'just in case'. Maybe you could put in the base64 images *just for the apps that you can test?*. A comment with the source filename for each would be great too - so if later it gets automated it won't be too hard to figure out which image is which?

m-p-3, 2022-11-21:
> Having the icons there as PNGs is great, but I'm a bit iffy about adding 95 icons 'just in case'. Maybe you could put in the base64 images *just for the apps that you can test?*. 

Of course, I wasn't planning to import them all I figured that since I was doing the work, I'd do more and have them ready to go if the demand for a specific app or service is high enough. I'll try to prioritize generic icons when it makes sense.

> A comment with the source filename for each would be great too - so if later it gets automated it won't be too hard to figure out which image is which?

Great idea, I can certainly do that.

Regarding the icon colour, I see there's another section to define a color. I'll do my best to include the official color of each app (from their press kit / brand guidelines) when possible, however I'm not sure how to easily test that to make sure it looks good, but that's on my end to figure out :)

gfwilliams, 2022-11-21:
> Regarding the icon colour,

Yes, totally - I wouldn't worry too much about that. I think a lot of times it can be more of a hinderance - sometimes a color works well on a light theme but not dark - hence there's the option to just turn that off :)

m-p-3, 2022-11-21:
Which is why I'd like to test it before I commit anything, however my programming skills are somewhat limited with the IDE. I'll figure it out :)

m-p-3, 2022-11-24:
I'll be at it shortly, just started a new job and I gotta prioritize a bit :)

m-p-3, 2022-12-01:
@gfwilliams Regarding the method used to select an icon, right now it's using the user-visible name when detecting the notification source. On the Android side of thing I think I found maybe a couple apps where there could be a name "collision" (NBC, the TV network and the National Bank of Canada using the NBC abbreviation as well). Also, I think I have some apps that change the user-visible name if you change the system locale.

Now I don't know how Apple works on their end with the ANCS or if it would make things even more complex on the backend, but would it be a good idea to identify an app notification source by relying on the AppID / Package name instead of the user-visible name?

Not sure what would be the best method of compiling that list of apps and associate them with their proper icons, without creating duplicate work.

gfwilliams, 2022-12-01:
It's a point - we could do it based on app ID if there's a problem.

Although at the end of the day it'd be quite a big change requiring Gadgetbridge mods - maybe wait and see if anyone really has a problem with this before we change it?

Right now, worth case they just don't get the right icon/color.

m-p-3, 2022-12-05:
I suppose I should have bumped the version and updated the changelog to see the change pushed through, my bad!

#2355 created.

gfwilliams, 2022-12-05:
Ah - sorry, I should have checked too :)

So do we think this issue is closed now?

rigrig, 2022-12-05:
Looks that way to me, thanks @m-p-3!

closed 2022-12-05 by rigrig