espruino/BangleApps #2196: Covert apps to use `require("widget_utils").hide()/show()` rather than duplicating code

gfwilliams, 2022-10-26:
Currently every app that wants to hide widgets does so with its own code. I just added a library and it'd be nice to change apps to use this.

Apps that look like they need changes right now are:

- [ ] apps/lcars/lcars.app.js
- [ ] apps/advcasio/app.js
- [ ] apps/neonx/neonx.app.js
- [ ] apps/pebbled/pebbled.app.js
- [ ] apps/notanalog/notanalog.app.js
- [ ] apps/aiclock/aiclock.app.js
- [ ] apps/macwatch2/app.js
- [ ] apps/limelight/limelight.app.js
- [ ] apps/cassioWatch/app.js
- [ ] apps/daisy/app.js
- [ ] apps/pebble/pebble.app.js
- [ ] apps/thering/app.js
- [ ] apps/rebble/rebble.app.js
- [ ] apps/imageclock/app.js
- [ ] apps/widviz/widget.js
- [ ] apps/bwclk/app.js
- [ ] apps/circlesclock/app.js
- [ ] apps/90sclk/app.js
- [ ] apps/widviztime/widget.js ?

Darin755, 2022-10-28:
Did you mean convert?

myxor, 2022-10-31:
Circlesclock is done with https://github.com/espruino/BangleApps/pull/2202

gfwilliams, 2022-11-23:
Just to add - for anyone else looking at this, widget_utils now contains `slideOn` which allows the widgets to be slid onto the screen with a downwards swipe

thyttan, 2022-12-01:
#2342 for aiclock

Rarder44, 2022-12-02:
#2227 for rebble

thyttan, 2023-02-07:
Maybe this can be closed since #2492 was merged?

gfwilliams, 2023-02-07:
Yes,great - thanks!

closed 2023-02-07 by gfwilliams