espruino/BangleApps #2206: gpstrek: Compass is inverted

nxdefiant, 2022-10-29:
Without GPS fix the compass shows inverted values. North is North, but e.g. east is west and vice versa.  The heading might need to be subtracted from 360 like e.g. in [compass](https://github.com/espruino/BangleApps/blob/master/apps/compass/compass.js#L52). Also I assume with GPS fix its heading is used? The documentation does not mention this. For lower speeds the GPS heading is unstable, so it might be a good idea to fallback to compass heading.

The other thing missing in the documentation is the "Slices"-option. What does it do?

Notify @halemmerich

halemmerich, 2022-10-30:
Yes, when available it uses the GPS course as heading. I have done some changes in https://github.com/espruino/BangleApps/pull/2210. The compass heading is also jittery, but i did not find an elegant way of filtering that to be better. Maybe a configurable threshold speed for switching between gps course and compass heading would be useful? I used this primarly for walking and had relatively good gps courses after some meters in a more or less straight line. At least better than the compass heading values.

Slices is/was just a bad name for the number of rows shown on each page.

As for the compass being inverted, that seems to be an error of firmware? I have added the same workaround as in the navigation compass app. @gfwilliams wasn't there some discussion on this a short while ago?

nxdefiant, 2022-10-30:
Guess I found the discussion: #1866.

A configurable threshold for compass/gps direction sounds good to me. Maybe in the future we could smooth it by combining both values in a kalman filter or similar sensor fusion algorithm. Downside is processing power required, not sure if it pays.

gfwilliams, 2022-10-31:
Thanks! Just merged https://github.com/espruino/BangleApps/pull/2210 to fix this

gfwilliams, 2022-10-31:
> As for the compass being inverted, that seems to be an error of firmware? I have added the same workaround as in the navigation compass app. @gfwilliams wasn't there some discussion on this a short while ago?

Yes, it's an odd one. Obviously there are apps that use this now, but I'm still kind of surprised it wasn't fixed or at minimum documented, but I just checked and the code there hasn't been changed in 3 years.

I guess the best bet would be to document this and *also* create a new field `navHeading` (any ideas on name?) that was correct, and to deprecate `heading`?

closed 2022-10-31 by gfwilliams

nxdefiant, 2022-11-01:
ok with new heading

halemmerich, 2022-11-01:
How about just fixing it in firmware and changing all 20 or so apps to using `(process.version.match(/.*v([\d]*)/)[1] > 15)?heading:(360-heading)` until 2v15 is old enough to remove that workaround? Alternatively the boot code could modify the 'mag' event for older firmwares.

gfwilliams, 2022-11-01:
My concern is that there are quite a few folks who don't put all their apps in the app store (including some companies), so I couldn't fix those... But I don't often break backwards compatibility and this seems like quite a minor thing to change.

So I'll put a change for `heading` in, and as you say, I'll get a fix in the bootloader for it on older firmwares.

gfwilliams, 2022-11-01:
Ok, all fixed. Apps & firmware

halemmerich, 2022-11-01:
Very nice!

nxdefiant, 2022-11-01:
@gfwilliams Looks like waypointer is now wrong, it does call Bangle.getCompass() instead of listening to 'mag'. Doesn't getCompass() need a fix in boot as well?

gfwilliams, 2022-11-02:
Argh, ok, thanks. Just fixed in bootloader 0.52