espruino/BangleApps #2212: gpstrek: Selected waypoint is always last

nxdefiant, 2022-10-31:
Steps to reproduce:
1. Open GPS Trekking
2. Swipe right to menu
3. Go to Waypoint
4. Select any waypoint

The selected waypoint is always the last one.

Looks like the error is in [showWaypointSelector()](https://github.com/espruino/BangleApps/blob/master/apps/gpstrek/app.js#L546):
```
  for (let c in waypoints){
    menu[waypoints[c].name] = function (){
      state.waypoint = waypoints[c];
      state.waypointIndex = c;
      state.route = null;
      removeMenu();
    };
  }
```

At the time of the execution of the anonymous function the loop iteration has already finished and the variable 'c' is always the last value.

Possible fix would be to call a function with parameter 'c' that returns a function:
```
  for (let c in waypoints){
    menu[waypoints[c].name] = (function(i) { return function (){
      state.waypoint = waypoints[i];
      state.waypointIndex = i;
      state.route = null;
      removeMenu();
    }})(c);
  }
```

@halemmerich Sorry, another one.

halemmerich, 2022-11-01:
Where is the effect of this? When I select a waypoint in the menu, the name of the waypoint and the distance in the target row  seem to change correctly for me.

halemmerich, 2022-11-01:
I have tried again with 2v15 and there the behaviour is as you described. @gfwilliams Were there some changes for https://github.com/espruino/Espruino/issues/2206 maybe?

nxdefiant, 2022-11-01:
I would never have thought this differs in firmware. Yes, I'm on 2v15, what are you running @halemmerich?

Also, what would be the correct behavior regarding js standards?

gfwilliams, 2022-11-02:
I'm pretty sure that this has never worked on Espruino (as I mentioned in that bug)... In Espruino `for (let c` behaves the same as `for (var c` with the exception of the scope of `c`.

@nxdefiant's fix would be the way forward - if you look it's basically what we do in all the other code that does stuff with menus like this (or use `waypoints.forEach((wp,c) => ...)` which might be cleaner if `waypoints` is an array)

> Also, what would be the correct behavior regarding js standards?

Best to discuss in https://github.com/espruino/Espruino/issues/2206 but if Espruino was handling `let` in a fully spec compliant way then I think the original code would work.

I can't say I'm desperately happy about another of one of these really odd JS quirks, but it is in the spec now :(

halemmerich, 2022-11-02:
Hm, it seems I have mixed up my gpstrek versions while testing this, there was one that had already changed the menu in a way that prevented this to happen on current firmware but not on 2v15 for whatever reason. Have now done a PR with a fix for this using the forEach approach: #2218

closed 2022-11-04 by nxdefiant