espruino/BangleApps #2223: [agpsdata] GPS fix is slow even after downloading the A-GPS data

micheledaros, 2022-11-03:
### Affected hardware version

Bangle 2

### Your firmware version

2v15

### The bug

### Description
After downloading the A-GPS data through the app `A-GPS Data Downloader App (v0.3)`,  GPS fix takes quite long.

### Steps to reproduce.
- connect the watch to a property configured GadgetBridge App, configured to allow internet access
- run the agps app, until you see the text "A-GPS Updated"
- go outside, and try to use any app that requires precise location (like run, or bike speedometer)
- notice that the GPS fix is slow (in some tests that I've done it took more than 20 minutes)
- I've tried different settings for `GNSS Type` (GPS, GPS+BDS+GLON)

### Expected Behavior 
- if I download the A-GPS by installing the `Assisted GPS Updater` app (`assistedgps`), I can get a GPS fix in a matter of seconds

micheledaros, 2022-11-03:
@pidajo or @myxor may have a hint :)

I'm more than happy to contribute, if I can be of any help

myxor, 2022-11-04:
Sometimes i experience the same and the only thing which speeds up the fix process is a reboot of the watch. I have no real idea why this happens or why the reboot helps.

micheledaros, 2022-11-04:
I tried rebooting several times, but in my case it did not help.

But actually, after a factory reset, and an update to the latest cutting edge firmware `2v15.90`, it started working \_(ツ)_/¯

gfwilliams, 2022-11-04:
Glad that's fixed it for you! But I do wonder why this happened in the first place. Maybe the GPS can get confused somehow (thinking it's halfway through a command or similar) and they won't accept the AGPS commands. It's going to be a hard one to track down.

Right now it seems we more or less just 'stream' the data non-stop to the GPS, and perhaps we can instead split it up and send it chunk by chunk and check if there's any response from the GPS itself to each block.

micheledaros, 2022-11-04:
IMHO one starting point to start investigating, is that running the commands from the browser (by using installing the `Assisted GPS Updater` app) worked reliably.

What's the difference between running the commands from the browser vs from the watch's app?

myxor, 2022-11-08:
@micheledaros can you please check the new version?

micheledaros, 2022-11-08:
> @micheledaros can you please check the new version?

I've tried twice, it works for me so far 🎉🎉

gfwilliams, 2022-11-09:
Excellent! Thanks for the fix @myxor !

closed 2022-11-09 by gfwilliams