espruino/BangleApps #2230: [Swipe menus] Not working on newer firmwares

thyttan, 2022-11-04:
### Affected hardware version

Bangle 2

### Your firmware version

2v15.90

### The bug

'[Swipe menus](https://banglejs.com/apps/?q=swscroll)' stopped working on newer firmwares. 

I will look at this and try to fix it sooner or later. I'll comment here when I begin. If someone else wants to look at this before I get to it please go ahead! :)

thyttan, 2022-11-05:
After some testing I found it was the custom bootloader I had installed that caused the bug. Using the Bootloader 0.52 from the official app loader solves the problem.

closed 2022-11-05 by thyttan

gfwilliams, 2022-11-07:
I think also the swipe menus were updated on friday so maybe you just didn't have those

thyttan, 2022-11-07:
> I think also the swipe menus were updated on friday so maybe you just didn't have those

Hmm, I don't think so. There's only the 0.01 version on the development app loader, which I committed September 22.