espruino/BangleApps #2241: [Slope Clock] Ram leak when loading launcher during animation

thyttan, 2022-11-06:
### Affected hardware version

Bangle 2

### Your firmware version

2v15.93

### The bug

Video showing the bug on this link: https://sprend.com/download?C=6ce80d7aee83402299727f2620d02e0f

Description of video:
1. On the clock RAM Widget will show 26% or 27% typically.
2. If the slope animation has finished and I load dtlaunch, the RAM Widget will say 13% on the launcher.
3. If I go back to Slope Clock and the switch back to launcher before the animation has finished, the RAM Widget will say 19% on the launcher.

The ram leak is consistent and fairly inconsequential, but it's there.

gfwilliams, 2022-11-07:
I'm actually really struggling to reproduce this with the default launcher, slope clock, and running `process.memory().usage` in the IDE console. Please could you try connecting with the IDE and running `process.memory().usage`? I'm wondering whether the widget just happens to update at the wrong time, and whether actually a few ms after it updates the extra memory usage has gone

thyttan, 2022-11-07:
On Slope clock when animation is finished:
```
>process.memory().usage
=2365
>process.memory().usage
=2365
```

On dtlaunch launched after the animation on Slope clock was finished:
```
>process.memory().usage
=1469
>process.memory().usage
=1469
```

On dtlaunch launched during the Slope clock animation:
```
>process.memory().usage
=2190
>process.memory().usage
=2190
```

On Slope clock during the animation:
```
>process.memory().usage
=3105
>process.memory().usage
=3105
```
I ran the command twice for every case to see if it was constant.

Before you do any more testing, I'll do a factory reset and install just the things you tested with.

thyttan, 2022-11-07:
> I'm actually really struggling to reproduce this with the default launcher, slope clock, and running process.memory().usage in the IDE console.

After a factory reset and installing Slope clock, Ram widget and Quick launch I don't get the ram leak either.

gfwilliams, 2022-11-07:
Thanks for checking that out after a factory reset! Yes, that's going to be a tricky one to track down... If you get this again (especially with stuff from the app loader) definitely let me know though!

thyttan, 2022-11-07:
Closing this since the problem doesn't seem to arise when using the official app loader.

thyttan, 2022-11-13:
I get this when only using stuff from official development app loader (+ my changes in open PR's) now as well, with the introduction of Bangle.load and the likes.

halemmerich, 2022-11-13:
The memory use does not seem to accumulate over multiple cycles and is removed by switching to slopeclock and letting the animation play out so it must be something only allocated during the animation. The only difference I could see is the drawing the date at the end so I tried setting another font (4x6) in the reset method (otherwise the last used font is the big PaytoneOne font). That seems to fix this issue **but** `g.reset()` does not while it should.

gfwilliams, 2022-11-14:
Thanks for tracking this down! I'll take a look this morning

closed 2022-11-14 by thyttan

gfwilliams, 2022-11-15:
Great - thanks for checking!