espruino/BangleApps #2246: [qmsched] Feature request: make scheduling quiet mode optional (like Theme and LCD settings)

jademonkee, 2022-11-08:
Hi there,
I'm looking for a way to set my watch to dark mode and dim the LCD during the night. Great that the 'Quiet Mode Scheduler' app (qmsched) does that, but I don't actually want it to set to quiet mode, too (I set it to dark mode and dim the LCD when my son goes to bed, but then it's several hours before I go to bed, and I'd like notifications during that time). When I do go to bed, I set my phone to DnD, which automatically sets my watch to DnD, so I don't need that managed in the app.
So I'm wondering if it would be possible to make the 'quiet mode' an optional setting in the app, like the Theme and LCD settings.
Or is there an existing way to achieve this that I'm not aware of?
Thanks.

gfwilliams, 2022-11-09:
Maybe @alessandrococco has some ideas here? As I understand it 'quiet mode' is just a set of options that is applied or not, so if those settings didn't include something that turned off buzzing, it'd work great

buildingleaver, 2022-11-24:
I have a similar need for a different use case: sometimes I wear earplugs at night and I want to rely on the watch to be woken up if someone calls. I really enjoy the fact that quiet mode allows to automatically dim the LCD and disable the 'wake on twist' setting, so having those features while keeping buzz on calls would be great.

rigrig, 2022-11-28:
Right now it works by
1. (old) schedule code that switches the global `quiet` setting
2. Every app that would buzz checks the setting to determine whether to stay quiet
3. At boot: if `quiet` is enabled, it overrides the theme/LCD settings

I think the app is in need of an overhaul: 
- update it to use the `sched` library
- change it to "Settings Schedule", i.e. don't hang it all on "quiet mode", but just allow scheduling any combination of DnD/theme/LCD setting changes