espruino/BangleApps #2247: LCD modes on Bangle.js 2

v1nc, 2022-11-08:
I am expecting to receive my Bangle.js 2 soon and want to port some old watch faces and create some new one.

Unfortunately I saw that the version 2 does not support LCD modes and `setLCDMode` does not work in the emulator.

Is this a hardware limitation or will this feature be added to version 2 in the future?

thyttan, 2022-11-08:
It's in [the Bangle.js 2 API reference](http://www.espruino.com/ReferenceBANGLEJS2#l_Bangle_setLCDMode), suggesting it is supported. Or is that a mistake?

v1nc, 2022-11-09:
> It's in [the Bangle.js 2 API reference](http://www.espruino.com/ReferenceBANGLEJS2#l_Bangle_setLCDMode), suggesting it is supported. Or is that a mistake? The pixel counts does not match Bangle.js 2 for example.

On the bottom of the description it says `This is only available in Bangle.js smartwatches`.
On other functions it says `This is only available in Bangle.js 2 smartwatches`, so I guess it is not supported.

gfwilliams, 2022-11-09:
Yes, I'm afraid it's not supported. On Bangle.js 1 it was kind of a hack to work around the LCD and not normally having an offscreen buffer, and that didn't seem to really be needed on Bangle.js 2 so I wanted to leave the complexity out.

What watch face were you looking at changing/what LCD mode/feature were you after? I can suggest how to work around it (or if you even need to)

v1nc, 2022-11-09:
Thanks for your reply.
At first I want to port the [mario watch face](https://github.com/espruino/BangleApps/tree/master/apps/marioclock).
I thought the LCD mode is handy for smooth animations, but now I noticed in the slope clock that there is an offscreen buffer for the version 2.

I also thought using the 80x80/120x120 mode would be good to develop version 1 and 2 compatible watch faces without differentiating between the different screen sizes.

v1nc, 2022-11-13:
Great, thanks for the advise :)

closed 2022-11-13 by v1nc