espruino/BangleApps #2258: Feature suggestion/idea: Edge to edge swipes reserved for global/system wide navigation

thyttan, 2022-11-12:
This is more of a brainstorm post than an actual suggestion - as of yet at least.

I like navigating with swipes, and love e.g.  the Quick Launch app. I think it could make sense to extend the idea of Quick Launch and Swipe to clock to be available no matter what app I am on, also if that app leans heavily on swipes for interactions. Then, a way to distinguish between swipes that should be picked up by the current foreground app, and ones that should fire global navigation would be if edge-to-edge swipes were reserved by this global navigation, be it in an app or firmware.

If I imagine there were one action per edge-to-edge swipe there could be at least up to 12 different 'slots' to fit functionality in. Here I count swipes between opposing sides, and between adjacent sides.

Please add to the brainstorm!

gfwilliams, 2022-11-14:
Right now we get a notification of swipes direct from the touchscreen controller, so we'd have to implement some special detection on top of what the touchscreen does.

However, I feel like we have the ability right now to do this with an extra app (eg pattern launcher or swipe to launcher) - I'm not sure there's really a reason to impose this on all users if they can just install an app that works they way they want?

thyttan, 2022-11-15:
Yes, I agree implementing it as an app makes more sense. I'll see if I ever get to it :P