espruino/BangleApps #2268: [Light Switch Widget] Featur request: Save brightness to setting.json

thyttan, 2022-11-13:
@storm64

I'd like it if Light Switch Widget updated the global brightness setting when releasing from sliding to change the brightness level. Maybe it would be best to add it as an option.

I tried implementing this, but it was harder than I expected.

This is what I tried to add at a couple of places:
```
        let storage = require("Storage");
        let settingJs = storage.readJSON("setting.js");
        settingJs.brightness = value;
        print(settingJs.brightness);
        storage.writeJSON("setting.js",settingJs);
```

storm64, 2022-11-13:
If I remember correctly the boot code is rewritten when a change in the settings.js is recognized. And to prevent this to happen quite frequently, I chose to use a separate position to store the brightness value.

thyttan, 2022-11-14:
Ah, I see. That makes sense. I'll close this as not planned.

closed 2022-11-14 by thyttan