espruino/BangleApps #2269: [swipe event] Quick swiping of the screen registers as touch event

thyttan, 2022-11-13:
### Affected hardware version

Bangle 2

### Your firmware version

2v15.136

### The bug

If doing a swipe on the screen too quickly it will be registered as a touch at the point the finger touched first. 

I'd like it if I could do faster swipes (maybe drags to) than I can today. 

One way to test this:
1. Start Desktop Launcher on the watch, make sure to have the 'Direct Launch' setting disabled.
2. Do a sufficiently fast swipe, aiming to start an up or down swipe motion with the finger on a currently not selected app icon.
3. If the swipe was quick enough the app will be selected (instead of the expected change of Desktop Launcher page).

I'll attach videos demonstrating the effect.

gfwilliams, 2022-11-14:
It would be good to see what you mean in a video...

But I'm afraid this is probably down to the touchscreen controller itself. I guess it's possible you may be able to configure it differently but right now Bangle.js gets its events direct from the controller, so it is that which decides what a swipe or touch is.

thyttan, 2022-11-14:
https://user-images.githubusercontent.com/97237430/201782412-7cf63e0d-d46c-4259-97c2-24cd0e36a997.mp4

It was admittedly a little hard to trigger this for the video, as you'll see. But it happens commonly enough for me if I'm not mindful to not be too quick when doing swipes, and that's the reason I don't have the "Direct launch" option set for Desktop Launcher.

gfwilliams, 2022-11-15:
It would be interesting if you could try `reset()` then `Bangle.on("drag",print);Bangle.on("swipe",print)` and do some quick swipes?

I just tried here and it looks like we just don't get the any real data points from the touchscreen if we swipe quick.

I think it's possible that the touchscreen isn't actually scanned *that* fast - maybe 50Hz or so - to save on power. It may be it's just not capable of detecting fast swipes.

thyttan, 2022-11-15:
Yes, seems like you're probably right. 

But it does seem like drag-event is a little quicker and can notice dx!=0 and dy!=0 even though a swipe isn't emitted. 

Extending your test code like this:
```
i = 0;
j = 0;
Bangle.on("drag",()=>{i++;print("drag "+i+":");});
Bangle.on("drag",print);
Bangle.on("swipe",()=>{print("\n");});
Bangle.on("swipe",()=>{j++;print("swipe "+j+":");});
Bangle.on("swipe",print);
Bangle.on("swipe",()=>{print("\n");});
```

I can get e.g. this output to console field:
```
drag 28:
{ "x": 103, "y": 139, "b": 1, "dx": 2,
  "dy": -13 }
drag 29:
{ "x": 107, "y": 118, "b": 1, "dx": 4,
  "dy": -21 }
drag 30:
{ "x": 107, "y": 118, "b": 0, "dx": 0,
  "dy": 0 }
drag 31:
{ "x": 107, "y": 110, "b": 1, "dx": 0,
  "dy": 0 }
drag 32:
{ "x": 108, "y": 104, "b": 1, "dx": 1,
  "dy": -6 }
drag 33:
{ "x": 108, "y": 104, "b": 0, "dx": 0,
  "dy": 0 }
```
Where I get some `dy<0`'s indicating we could do something with that.

Maybe the swipes filter the input like this intentionally to avoid emitting swipes when it probably shouldn't?

gfwilliams, 2022-11-15:
Yeah, that's what I'm seeing too.

> Maybe the swipes filter the input like this intentionally to avoid emitting swipes when it probably shouldn't?

Yes...You're seeing DY of 13 and 21 - so 34px movement in total (not much more than the height of the widget bar). I think it's fair that the touchscreen discounts that, or you'd end up 'swiping' whenever you went to just press a button and rocked your finger slightly.

I'm afraid unless you can find some magic way to reconfigure the touchscreen (it's the same one as in PineTime so maybe there is some info out there) you're out of luck

thyttan, 2022-11-15:
Ok, thanks for helping to test this! I'll close this issue as not planned. If I find something interesting regarding this I might open it up again.

closed 2022-11-15 by thyttan