espruino/BangleApps #2287: Torch v0.10: Fails to exit

nxdefiant, 2022-11-17:
On BangleJS2 with firmware 2v15.93 I can no longer exit the Torch application by clicking the hardware button. In the console I get:

> Uncaught Error: Function "showClock" not found!
 at line 1 col 543 in torch.app.js
Bangle.showClock()
       ^
in function "btn" called from line 1 col 8
b.btn(1)
       ^
in function called from system

nxdefiant, 2022-11-17:
I'll try a firmware update in the evenning.

halemmerich, 2022-11-17:
Maybe your boot files are not up to date? You could try Settings->Rewrite Settings. Then there should be polyfills for these methods if your firmware is older than 2v15.135.

gfwilliams, 2022-11-17:
I think you just haven't updated the bootloader in the app loader? Make sure you're on the latest version and try again.

gfwilliams, 2022-11-17:
... if you had updated and 'rewrite settings' helps I'd love to know - it might be we need to update the bootloader such that it causes an automatic rewrite

nxdefiant, 2022-11-17:
According to the app loader the bootloader is up to date v0.53. Rewrite settings did not wake, a firmware update to .16 did.

nxdefiant, 2022-11-17:
I guess [bootupdate.js#L86](https://github.com/espruino/BangleApps/blob/master/apps/boot/bootupdate.js#L86) might fail since it compares
`215.93 < 215.135` and not `215.093 < 215.135` 

Always having 3 digits as version suffix could help next time.

gfwilliams, 2022-11-17:
Thanks - yes, that could do it. There's actually no reason for that version check anyway so I'll pull it out completely.

Version compares are a pain - we need a better standardised way of doing it really - it's a bit late to adopt 3 digit codes now :)

closed 2022-11-18 by nxdefiant