espruino/BangleApps #2295: Calendar does show Widgets when started from QuickLaunch

nxdefiant, 2022-11-20:
With firmware 2v16, QuickLaunch v0.06 and Calendar v0.11 on BangleJs2:

The new version of QuickLaunch uses Bangle.load() to start applications even if the started application does not have any space left for widgets, e.g. Calendar.

How to reproduce:
1. Start Calendar from QuickLaunch
2. Wait until the display backlight turns off
3. Press the button to turn the backlight on again

You now see widgets on top of the calendar application.

This behavior is as [documented](http://www.espruino.com/Reference#l_Bangle_load):
> This should only be used if the app being loaded also uses widgets

The question is: What we do now?
a) Fork QuickLaunch to a new version that uses the old load()?
b) Add a settings option whether to use the new Bangle.load() or global load()?
c) Add the information if the app uses widgets to metadata and use that to decide in what way to load it?
d) ...?

nxdefiant, 2022-11-20:
Notify @thyttan

halemmerich, 2022-11-20:
Can you try out if [https://halemmerich.github.io/BangleApps/?id=widgetupdate](https://halemmerich.github.io/BangleApps/?id=widgetupdate) helps you?
That could be a possible d). It keeps widgets visible/hidden state and per default hides widgets on `Bangle.load`.
Another alternative would be building a list of apps that stores if it contains `Bangle.loadWidgets` just as `.bootcde` does for loading the current clock.

thyttan, 2022-11-20:
Yes, this is my fault for adding Bangle.load prematurely. I'll do a PR for reverting it. Thanks for reminding me! 👍

nxdefiant, 2022-11-20:
@halemmerich It basically seems to work. Side effect: alarm & timer eta widget does not display show before the display is locked.

halemmerich, 2022-11-20:
That's interesting, I did not know this widget. Anyway a neat idea for a widget, will have to try it out and see why it fails to display correctly.

gfwilliams, 2022-11-21:
Just merged, so I think this is sorted now

closed 2022-11-21 by gfwilliams