espruino/BangleApps #2307: truncated uploads with Util.writeStorage

wagnerf42, 2022-11-22:
### Affected hardware version

Bangle 2

### Your firmware version

2v15

### The bug

file upload still truncates large files.

Util.writeStorage is still truncating large files (i got again some problems with truncated files on gipy)

so i added a test after the upload by redownloading and comparing uploaded content.
you can try the test on the new interface here : https://wagnerf42.github.io/BangleApps/

i also added a test on gipy itself to avoid starting biking with incomplete files

wagnerf42, 2022-11-24:
hi,

if you take this file https://www.cirkwi.com/fr/exporter-gpx/67942_0?mb_use_temp=1
(full description is here https://www.cirkwi.com/fr/circuit/67942-la-chapelle)

if you go to the latest version of gipy  https://wagnerf42.github.io/BangleApps/
and you disable 'fetch from openstreetmap'.
this will generate a 301 bytes gpc file.
when uploaded only 299 bytes go to the watch and the interface.html should tell you the upload failed.
you can double check that by launching gipy (latest version) on the watch and loading the file:
it will compute the expected file size using the headers and complain that 2 bytes are missing.

Fred

gfwilliams, 2022-11-24:
For future ref, the data written is:

```
[130, 185, 3, 0, 18, 0, 0, 0, 0, 0, 5, 105, 198, 162, 233, 172, 23, 64, 232, 77, 69, 42, 140, 173, 70, 64, 232, 159, 224, 98, 69, 173, 23, 64, 118, 137, 234, 173, 129, 173, 70, 64, 144, 136, 41, 145, 68, 175, 23, 64, 40, 155, 114, 133, 119, 173, 70, 64, 117, 171, 231, 164, 247, 173, 23, 64, 112, 153, 211, 101, 49, 173, 70, 64, 177, 167, 29, 254, 154, 172, 23, 64, 69, 158, 36, 93, 51, 173, 70, 64, 207, 218, 109, 23, 154, 171, 23, 64, 55, 253, 217, 143, 20, 173, 70, 64, 36, 156, 22, 188, 232, 171, 23, 64, 248, 170, 149, 9, 191, 172, 70, 64, 234, 33, 26, 221, 65, 172, 23, 64, 41, 203, 16, 199, 186, 172, 70, 64, 177, 225, 233, 149, 178, 172, 23, 64, 113, 61, 10, 215, 163, 172, 70, 64, 229, 237, 8, 167, 5, 175, 23, 64, 177, 138, 55, 50, 143, 172, 70, 64, 58, 175, 177, 75, 84, 175, 23, 64, 134, 201, 84, 193, 168, 172, 70, 64, 30, 138, 2, 125, 34, 175, 23, 64, 254, 38, 20, 34, 224, 172, 70, 64, 60, 165, 131, 245, 127, 174, 23, 64, 34, 84, 169, 217, 3, 173, 70, 64, 144, 102, 44, 154, 206, 174, 23, 64, 126, 169, 159, 55, 21, 173, 70, 64, 89, 110, 105, 53, 36, 174, 23, 64, 254, 241, 94, 181, 50, 173, 70, 64, 229, 155, 109, 110, 76, 175, 23, 64, 111, 71, 56, 45, 120, 173, 70, 64, 231, 53, 118, 137, 234, 173, 23, 64, 175, 206, 49, 32, 123, 173, 70, 64, 148, 222, 55, 190, 246, 172, 23, 64, 232, 77, 69, 42, 140, 173, 70, 64, 5, 192, 2].length
```

BUT it looks like your version of `core` (AppLoaderCore) is the out of date version from before I fixed this.

If you update your repo this'll be fixed

closed 2022-11-24 by gfwilliams

wagnerf42, 2022-11-24:
oh, sorry for that.

i'm actually not familiar with git submodules so i assumed it followed the latest version automatically.
(since i don't need to checkout the submodules to generate the web pages).

thanks for the help.

gfwilliams, 2022-11-24:
No problem! Git submodules can be tricky - but I think if you can merge all upstream changes to the app loader into your app loader then the submodule should update automatically?

wagnerf42, 2022-11-24:
yes, i can confirm, it's all ok now.

thanks