espruino/BangleApps #2318: health: global step count target

gfwilliams, 2022-11-28:
It looks like `wpedom` and `health` both have their own step count settings - really we should make this global.

If not in `health` maybe it should even be in global settings.

Comment here: https://github.com/espruino/BangleApps/pull/2314#discussion_r1033373999

myxor, 2022-12-01:
Good idea. Also i would like a notification when the daily step goal is reached (we should be able to toggle this).

glemco, 2022-12-06:
I'm not really fond of the `health` app, partly because I'm not interested in those fine grain graphs on the watch (which I can somehow see in gadgetbridge already), while I'd rather see some punctual stats (steps today and hearthbeats without too many taps).
I'd support having this setting as global, so that we wouldn't require users to have `health` installed at all.
Bonus feature: not sure if GB supports the synchronization of it, but it has already a daily steps goal. It would be nice to have it uploaded to the bangle!

bobrippling, 2023-11-15:
I guess this wants to stay open still? There's a few other apps that count steps themselves:
- [widpedom](https://github.com/espruino/BangleApps/blob/98e560a75649/apps/widpedom/widget.js#L20-L41)
- [banglebridge](https://github.com/espruino/BangleApps/blob/98e560a75649/apps/banglebridge/widget.js#L200-L203)
- [gpstrek](https://github.com/espruino/BangleApps/blob/98e560a75649/apps/gpstrek/widget.js#L102-L104)
- [activepedom](https://github.com/espruino/BangleApps/blob/98e560a75649/apps/activepedom/widget.js#L215-L215)

gfwilliams, 2023-11-16:
Thanks! Yes, it'd be really good to just use `Bangle.getHealthStatus("day").steps` where possible - only downside is it forgets after the watch does a full reboot (it could be fixed in firmware at some point), but I feel like hard reboots should be super rare for pretty much everyone now