espruino/BangleApps #2353: alarm: Missing option to hide widget

nxdefiant, 2022-12-04:
The alarm widget is missing an option to globally disable the "bell"-widget. Now before we add a settings file for alarm:

Shouldn't the widget be moved to the sched library? Any vote against?

gfwilliams, 2022-12-05:
> Shouldn't the widget be moved to the sched library? Any vote against?

I'd vote against this - someone will want to implement their own alarm app/widget and they won't want the default one installed.

And having spent quite some time trying to remove the duplication in 7 different alarm/timer apps and put the code in `sched` the last thing I want is someont trying to fork `sched` :)

I'd argue maybe if you really don't like the widget in the alarm app (why?) maybe it should be its own widget, and then users could just uninstall it if they don't want it showing?

nxdefiant, 2022-12-05:
Why I want to move the alarm widget from alarm to sched: Because it uses code from sched.
Also sched already has a settings page, so I would not need to create one to hide the widget.

Why I want to hide the widget? Because I have widalarmeta installed which already indicates an alarm. Actually one alternative I have thought about was to add code to widalarmeta to hide the alarm widget..

What I dislike about moving the alarm widget into a separate widget app is that unless we make it a dependency of alarm it will not be installed automatically.

But I would also be ok with giving alarm a settings page with the option to hide the widget.

nxdefiant, 2022-12-05:
Update: Maybe I'll edit widalarmeta to hide the alarm widget anyway. widalarmeta only displays an alarm in less then 24h.
So for < 24h I could hide the alarm widget and for > 24h not.

nxdefiant, 2022-12-05:
dependencies & provides_modules sounds like a plan.

nxdefiant, 2022-12-05:
Some problems with the module approach:
While provides_modules worked in my testing the linter correctly complains that a widget is not a module:

https://github.com/nxdefiant/BangleApps/commit/14acbe6f6c40151b9a9d52bc71c5b388f0f751dc#diff-7f3af777f14b61163f5ddc5b0e3aa40be3aa9ff6f44cce3e66e1bc7b791c44a2

Also I would like a way to configure "widalarm" as the default "alarmwidget" module. The current mechanism seem to sort alphabetically so for now "widalarm" is preferred over widalarmeta, but this does seem to be guaranteed.

gfwilliams, 2022-12-06:
Yes - I guess I could modify things to have a `provides_widget` - or I  guess there could be an empty module that gets added (or maybe one with a `hide` method)

> Also I would like a way to configure "widalarm" as the default "alarmwidget" module.

We normally use the sortorder - so if it is set it -1 then it always gets put first. If you want to modify the code to add a default option then that could be a good idea though.

nxdefiant, 2022-12-06:
I would prefer the provides_widget, it sounds 'cleaner'.

gfwilliams, 2022-12-06:
Ok, sorted

closed 2022-12-06 by gfwilliams