espruino/BangleApps #2361: [BW Clock]  hangs and locks when sunrise_clockinfo is installed.

hughbarney, 2022-12-05:
### Affected hardware version

Bangle 2

### Your firmware version

2.16.7

### The bug

CONFIGURATION:
Bangle 2.
Firmware 2.16.7
BW clock 0.23
Slopeclock++ 0.5
Sunrise Info Clock 0.1
Latest everything else.

Steps to reproduce

install slopclock++ and observe that sunrise and sunset options are not displated in the clock info menu's
install BW clock and scroll through the Bangle menu without issue
install sunrise_clockinfo and observe sunrise and sunset options now in the clock info menus.
start up BW clock and select the Bangle menu. At some point it will hang.
You will see the following on the IDE console:

Uncaught Error: Unhandled promise rejection: undefined
in function "altUpdateHandler" called from line 1 col 2110 in bwclk.app.js
...;alt="--";altUpdateHandler();
                              ^
in function "show" called from line 29 col 190 in bwclk.app.js
...ings.menuPosY-1];item.show();
                              ^
in function "drawMenuAndTime" called from line 38 col 509 in bwclk.app.js
....length+1);drawMenuAndTime();}if(is_upper){if(e.y<widget_size)...
                                ^
in function called from system
>
You will not see a sunrise or sunset option in the Bangle Menu.
Thereafter a black bottom half of the screen is displayed whenever you load BW clock.

peerdavid, 2022-12-06:
I wonder if this clock_info works in slopeclock++ when installed? seemsike a bug in sunrise to me...

Still BW clock should not hang I will lokk into this :)

gfwilliams, 2022-12-06:
I just checked and this works fine in slopeclock++

**I can also reproduce this exact error in BW Clock without the sunrise widget** - just right after starting the clock, keep tapping really quickly, and then after a few taps it stops updating.

In the stack trace posted initially it sounds like the error is from altUpdateHandler - and I wonder whether the issue is actually that when flicking through items quickly, altUpdateHandler calls the 'redraw' event a second or so late but the redraw handler hasn't been removed even though the altitude has since been hidden? 

Interestingly I also get a different error in BW Clock sometimes:

```
Uncaught Error: clearInterval(undefined) not allowed. Use clearInterval() instead.
 at line 1 col 2449 in bwclk.app.js
clearInterval(this.interval),delete this.interval
                                    ^
in function "hide" called from line 22 col 81 in bwclk.app.js
item.hide();lock_input=false;var info=item.get();drawMenuItem(info....
                 ^
in function called from system
```

And in that case, it looks like `.hide` is being called more than once on the clock_info item. IMO we shouldn't really have to cope with that for every clock_info implementation (then we start having to worry about `show` being called more than once too), so ideally the navigation of clock_infos would be smart enough not to do it.

... When this gets fixed it would be really good to bring the navigation code into a common library - afaik the code is now in another clock face too? So we'd have to fix that one as well.

thyttan, 2022-12-25:
>Uncaught Error: Unhandled promise rejection: undefined
in function "altUpdateHandler" called from line 1 col 2110 in bwclk.app.js
...;alt="--";altUpdateHandler();

Maybe this will be fixed with PR #2424? It modifies altUpdateHandler to catch an error if Bangle.getPressure() returns `undefined`.

thyttan, 2022-12-25:
If it's alright I'll add a bug report as a comment here.

Bangle 2 

fw 2v16.45

BW Clock stops updating graphics and stops taking touch input for me when:
- trying to display week number on the bwItems menu.
- trying to navigate up/down on the Weather menu (I do not have the weather app set up to get the weather but the app is installed)
- trying to navigate up/down on the Alarms menu.
- trying to navigate up/down on the Timer menu.

I can however do swipes for launching apps with Quick Launch or open the launcher by pressing the hardware button. Touching the Light Switch Widget will still work as well. So the bangle doesn't hang completely.

When I then launch BW Clock again via the launcher the bottom part of the app will not display, only the date field, and it still won't take input. I can resolve this by setting `"menuPosX":0,"menuPosY":0` in bwclk.setting.json before launching BW clock via the launcher.

BartS23, 2022-12-28:
The error occurs when a clock_info module does not call `redraw` in the `show` method.

In line 295 lock_input is set to true and `item.show` is called.
https://github.com/espruino/BangleApps/blob/1f1b8a20d566eddedd397d06af9e578b1547ce81/apps/bwclk/app.js#L295-L297 

The only way to get it to false is Line 137 in `drawItem` method
https://github.com/espruino/BangleApps/blob/1f1b8a20d566eddedd397d06af9e578b1547ce81/apps/bwclk/app.js#L136-L137

`drawItem` is an eventhandler for redraw
https://github.com/espruino/BangleApps/blob/1f1b8a20d566eddedd397d06af9e578b1547ce81/apps/bwclk/app.js#L143

thyttan, 2022-12-28:
@BartS23 Ok, thanks! I tried reverting commit https://github.com/espruino/BangleApps/commit/8425f0b50aea39a9083b95ccc9f4211d9c5d556a and with that BW Clock works fine again for me. Can be tested by reinstalling all apps via [my app loader](https://thyttan.github.io/BangleApps/?q=bw).

So either that commit should be reverted or further fixes are needed.

gfwilliams, 2023-01-05:
Thanks for checking up on this - I think it's probably best to keep https://github.com/espruino/BangleApps/commit/8425f0b50aea39a9083b95ccc9f4211d9c5d556a and change BWClock/etc?  Requiring the redraw in `show` seems like it's probably going to bite us later if someone else commits a new clock_info