espruino/BangleApps #2370: [messages] file write optimisation #2

gfwilliams, 2022-12-07:
Originally mentioned at https://github.com/espruino/BangleApps/issues/1460 - but Storage file access has really improved since that issue (plus fast load) and it seems the StorageFile optimisation probably isn't the best choice any more.

**When a message is received, we want to avoid writing to Storage as much as possible**

- [ ] When we receive a message and 'fast load' to the messagesgui app, we should be able to avoid a write+read from messages (we'll still need to be sure to save when we leave the app)
- [ ]  Just delete the messages file instead of writing `[]` (mentioned at https://github.com/espruino/BangleApps/pull/1498#issuecomment-1340912510)
- [ ] Messages app should be able to fast-load back to the clock, and if so it could maybe avoid writing messages then too (but ensuring we write an update when we leave the clock again could be an issue)

In the most basic case:

We're in a clock, we get a message and that message is dismissed, and we go back to the clock

We should be able to avoid *any* writes...

rigrig, 2022-12-07:
Also:
* [ ] Messages app should not rewrite the file on every change, but only on exit. It already does `E.on("kill", saveMessages);`, so I *think* we can just delete all other calls to to `saveMessages()`.

<br>

> * [ ] Messages app should be able to fast-load back to the clock, and if so it could maybe avoid writing messages then too (but ensuring we write an update when we leave the clock again could be an issue)

Writing messages from `E.on("kill")` should work, but keeping them around even after the UI is supposedly unloaded might give `MEMORY` problems? (especially on Bangle.js 1)

gfwilliams, 2022-12-07:
> keeping them around even after the UI is supposedly unloaded might give MEMORY problems? (especially on Bangle.js 1)

Yes, that's true... I'll take that one off - I don't think it would be a huge improvement anyway