espruino/BangleApps #2382: Settings / Device Hangs when scrolling up and down through long menus

hughbarney, 2022-12-11:
### Affected hardware version

Bangle 2

### Your firmware version

2.16.28

### The bug

Whilst scrolling up and down through the settings menus the Bangle2 hangs and becomes unresponsive.
When connected to the IDE there are no error messages.
The only way to recover is a long press reboot of the Bangle 2.

I been experiencing this kind of hang on all of the v2.16.28 builds.

I have narrowed it down to a minimum configuration of 
settings 

... checking a few things, will come back and update the defect.

hughbarney, 2022-12-11:
Attached id a backup file of config.
[issue_2382_banglejs_backup.zip](https://github.com/espruino/BangleApps/files/10201996/issue_2382_banglejs_backup.zip)

thyttan, 2022-12-11:
>Attached id a backup file of config.
[issue_2382_banglejs_backup.zip](https://github.com/espruino/BangleApps/files/10201996/issue_2382_banglejs_backup.zip)

I restored the backup but I haven't managed to reproduce the bug.

hughbarney, 2022-12-11:
I have reproduced quite easily on a 2nd Bangle 2 now.
Maybe I need to make a short video.

thyttan, 2022-12-11:
>I also noted that installing swipe menus made the issue go away, then I could uninstall swipe menu and it would come back.

This really makes me think it's something with `E.showscroller` that isn't updated with the latest changes, as suggested on [the forum thread](https://forum.espruino.com/comments/16794331/).

BartS23, 2022-12-11:
I was able to reproduce the bug.

I have narrowed it down to the following:
```
var menu = {
  '< Back': () => print
};

for (n = 0; n <= 10; n++) {
  menu[n] = {};
}

E.showMenu(menu);
```

If you enter the following lines in the left side of the IDE, the watch will hang/reboot
```
Bangle.emit("drag", { "x": 0, "y": 0, "b": 1, "dx": 0, "dy": -250 })
Bangle.emit("drag", { "x": 0, "y": 0, "b": 1, "dx": 0, "dy": 167 })
```

The bug occurs only when Back is set.

gfwilliams, 2022-12-12:
Thanks for the report - and thanks @BartS23 for such a concise way to reproduce this - I'll have a look and see if I can find/fix this this morning.

Just to add I managed to reproduce with the standard launcher too (as you'd expect given showScroller). Given the hang it may be related to the scrolling code - it's possible it's related to the LCD speed improvements that went into 2v16

closed 2022-12-12 by gfwilliams

hughbarney, 2022-12-12:
I have tested and the  hang bug is fixed in 2.16.33 but I still get screen corruption in the settings app on scrolling through long menus.  I will log a new bug for that.