espruino/BangleApps #2386: [messageicons] Short description of bug

rigrig, 2022-12-11:
### Affected hardware version

Bangle 1, Bangle 2

### Your firmware version

2v16.26

### The bug

Some icons (like `whatsapp`) now include a background color instead of being transparent. This causes issues when they are shown on a coloured background.

Before: 
[image]

After: 
[image]

Looks like this broke with [this commit](https://github.com/espruino/BangleApps/commit/938f881653fed9c3e81a6a1b6785bdde81e11f9b#diff-65b3b2b96b0abcc9eebaaec68d73354765036aab7f9fe7296c53edc1c3abb5caR71)

rigrig, 2022-12-11:
The images under `/icons` do have the right transparency, so maybe we should not spend too much time on manually fixing images, but look into writing some tool to export them properly instead?

gfwilliams, 2022-12-12:
> maybe we should not spend too much time on manually fixing images, but look into writing some tool to export them properly instead?

True. I've just added something that should fix this and make it faster/easier to add more icons too.

gfwilliams, 2022-12-12:
Fixed in https://github.com/espruino/BangleApps/commit/6a434f35ccf3aa8fa2254c221dab731e238c426a

closed 2022-12-12 by gfwilliams

m-p-3, 2022-12-13:
Thanks for improving this, that looks quite neat code-wise :)

gfwilliams, 2022-12-13:
No problem - should be a lot faster too. If anyone wanted to copy the colors into the JSON then I'd be happy to have a go at making it auto-generate that function as well!