espruino/BangleApps #2392: [weather] Clkinfo `conditionWithTemperature` item not showing the range

glemco, 2022-12-12:
### Affected hardware version

Bangle 2

### Your firmware version

2v16

### The bug

Since the last update to the weather clkinfo (0.22), a new item in the weather `conditionWithTemperature` has been created, now this item cannot be viewed as ranged item in clocks like circleclock as it is missing the proper setting (`hasRange`).
The solution is trivial, however I'd like to ask a more broad question about this specific item, I suppose the item was created to represent the weather circle in clkinfo (and export the feature to other clocks), the idea is good but the nice stuff about that circle was the ability to show multiple data in the circle (e.g. the range could be set to humidity as the temperature is already available as text).
To do the same with this item we could:
* Create a lot of items like `conditionWithXYZ` for all useful combinations of texts and ranges (just kidding, this is clearly not ideal)
* Have a setting somewhere for this (requires a few changes in the clkinfo module but I have the feeling this is what will happen sooner or later)
* Change this specific item to show the humidity in the range (forgetting about wind, which was the other option in circlesclock, temperature wasn't available in that gauge) and maybe call it `conditionWithData`, waiting for a possibility to set it up

Now, the 3rd option was my usecase (purely personal preference, but quick and easy fix). While adding the `hasRange` field, I would also modify that, to add more details to the clkinfo.
Again, that boils down to personal preference, so I would wait for some opinion before changing that.

bobrippling, 2023-11-15:
I believe we have [the `conditionWithData`](https://github.com/espruino/BangleApps/blob/98e560a756492303fd51c7a7daddc9b4fa3c3616/apps/weather/clkinfo.js#L42-L52) part now, but I don't quite follow the extra details - is there more to this issue would you say, @glemco?

glemco, 2023-11-20:
My initial idea was to have what was supported by circlesclock before clkinfo. Basically a circle could show the condition, a number/text (e.g. temperature) and an indicator/gauge (e.g. relative humidity).
Currently conditionWithData would show temperature for both the text and the numerical value (indicator). It could have been nice to be able to configure what is shown to make different combinations.
I'm not sure how many other clocks would benefit from having text and value reporting different data, but in any case the only sane way to implement this would be to have the possibility to configure the clkinfo, which I don't think is possible at the moment.
Does it sound a little clearer? The basic functionality is already good and there's probably no real need to make it much more complicated, I opened the issue in case someone else was interested to discuss the functionality, but perhaps at this point it could also be closed.

bobrippling, 2023-11-22:
Gotya, if I follow correctly, you're saying we could modify clkinfo to be more configurable, so a user can pick what text is shown against what kind of image/gauge/value? Thanks for taking me through the idea - I think we can leave this open if someone wants to have a go at it in that case