espruino/BangleApps #2395: [lcars/advcasio/cassioWatch/entonclk] don't report heart rate properly

gfwilliams, 2022-12-14:
It seems many clocks report HRM and use this pattern:

```
Math.round(Bangle.getHealthStatus("last").bpm)
```

But this returns the best HRM in the last 10 minutes - so it won't update for between 0-10 minutes.

Probably what's needed is really:

```
Math.round(Bangle.getHealthStatus().bpm||Bangle.getHealthStatus("last").bpm)
```

To get the 'best' HRM reading from this 10 minutes, or the reading from the last 10 minutes, or 0... and maybe even updating based on the HRM event.

But the best way would be to use the new `clock_info` module instead as this handles it properly now, and we can keep it updated if we spot issues like this in the future.

Affected:

- [ ] `lcars`
- [ ] `advcasio`
- [ ] `cassioWatch`
- [ ] `entonclk`

amunizp, 2022-12-27:
I have lcars in bangejs 2. HRM is stuck at 0.

amunizp, 2023-05-19:
@peerdavid and @Ronin0000 I don't know how to do clockinfo but copy and pasted patch.

closed 2023-05-24 by gfwilliams