espruino/BangleApps #2398: [messagegui] Back button not working on fast loaded notifications

Kedlub, 2022-12-14:
### Affected hardware version

Bangle 2

### Your firmware version

2v16

### The bug

The back button (and physical button) in a notification is not working after the latest update. It seems only to happen when the notification is fast-loaded while on clock.

### Expected behavior
A back button that would return to the notification list

### How to reproduce
While on the clock, receive a notification. Observe that the back button is not working

gfwilliams, 2022-12-14:
Ok if you try the development app loader and update then I *think* this will be fixed.

@rigrig is the `.show` stuff really needed? As far as I remember the messages app just showed the last unread message in the message list - which seemed like it worked fine.

closed 2022-12-14 by gfwilliams

rigrig, 2022-12-14:
@gfwilliams It's used [here](https://github.com/espruino/BangleApps/commit/2a506e7421f2320ba65afcadeb9d3458a8958dd1#diff-d07e6232280cd3e93fc023c3eccf06424471b22ea37c4eccac390beebf523a3d) so `messagesmusic` can just tell the UI to open the music screen, instead of fiddling with `settings.openMusic`.

To be honest I expected more uses when I added that option, but I don't really see any, so we could probably get rid of it.

gfwilliams, 2022-12-15:
Ahh, interesting thanks! I guess in that case we could just ensure that the music message is new and is the first one in the list? I'll have a look at that now

gfwilliams, 2022-12-15:
Ok, just pushed some changes - hopefully that all works ok. It seems to work fine for me now.

It seemed like you could do 'messages.openGUI(...)` without the message even being saved in some cases, so hopefully that helps!