espruino/BangleApps #2409: Lock Widget 0.07 , not disappearing when BTN is pressed

hughbarney, 2022-12-17:
### Affected hardware version

Bangle 2

### Your firmware version

2v16.24

### The bug

Allow the screen to lock

Now click the button
The LCD will come on but the lock icon does not clear.

Configuration:
  FW 2.16.34
  Bootloader 0.55
  Settings 0.55
  Launcher 0.20
  Anton 0.11
  Lock Widget 0.07

gfwilliams, 2022-12-19:
Do you see anything on the Web IDE if you connect and try and unlock?

Those are basically the factory apps and I have a watch here with them running, and it works fine.

So I wonder whether something has got broken on your watch? It might explain https://github.com/espruino/BangleApps/issues/2413 as well

hughbarney, 2022-12-19:
Can reproduce on 2.16.45 as well.
The issues goes away if you install another widget.  Eg the standard battery widget.
Will check the IDE now.

gfwilliams, 2022-12-19:
> So seems a problem when widlock is the ONLY widget.

Ahh, right! So it's basically that when the widget is hidden, it takes no space and drawWidgets thinks there is no widget bar. Fixed now - more info in https://github.com/espruino/BangleApps/issues/2413#issuecomment-1357430792

commit 39418b43 (2022-12-19): 0.08: Don't completely remove the lock widget when screen unlocked (use 1px) to ensure appRect/drawWidgets still thinks there are widgets (fix #2409, #2413)
closed 2022-12-19 by gfwilliams