espruino/BangleApps #2413: LCD Clock - not handling widgets properly

hughbarney, 2022-12-18:
### Affected hardware version

Bangle 2

### Your firmware version

2v16.45

### The bug

The LCD clock does not appear to be handling widgets properly.

There are a few disolay issues:
1. The initial display has the PM/AM DAY / DATE field overlap the bar in the middle
2. When the lock screen kicks in the top of the 2 squares at the top are overwritten.
3. When Firmware Clock Info is used the font is reduced to such a point that it cant be read.
It looks like the width is being estimated on the length of the string rather than using

gfwilliams, 2022-12-19:
What apps do you have installed? Do you see any errors in the IDE?

Because I just installed this on a watch again here and it work fine

gfwilliams, 2022-12-19:
> When Firmware Clock Info ...

You can check in the code yourself but it does get the string width properly. It's just that with the font it has, you're limited to either 1x or 2x scaling - and maybe 1x is too small to be easily visible.

hughbarney, 2022-12-19:
Issue goes away when a second widget is installed.  Seems related to #2409.

gfwilliams, 2022-12-19:
Ok, this is a tricky one. I think you will find that pretty much every single app using appRect fails like this if you have *just* the lock widget installed.

I've just 'fixed' this by making the lock widget take up 1px even when it's not shown. It seems a shame to have to do this when it's a very rare case though, and I think you in fact campaigned for `widlockunlock` so you probably wouldn't even use this anyway?

closed 2022-12-19 by gfwilliams