espruino/BangleApps #2435: Minor bug in health app

notEvil, 2022-12-29:
Hi,

this timeout
https://github.com/espruino/BangleApps/blob/1f1b8a20d566eddedd397d06af9e578b1547ce81/apps/health/boot.js#L14
seems way too long. Shouldn't it be just 60k ms?

myxor, 2023-01-03:
Pinging @halemmerich because it looks like these lines of code are from him :)

halemmerich, 2023-01-03:
@notEvil I think you are correct, the inner timeout only needs 60k ms timeout.

For reference the explanation of what this should have done:
This is for the 3 minute setting. In this case it needs to split the 10 minutes of each health event into 3 parts using two timeouts (and two timeouts to cleanup a minute later). The 3 minutes are actually 3:20 because of this 3 way split of 10 minutes.
So one timeout switching HRM on after 3:20min = (180 + 20)k ms and one switching off 60k ms later than that. The same for 6:40min as second run of the loop.

gfwilliams, 2023-01-05:
Thanks - yes, this looks like a bug.

I guess it was expected that the 2 setTimeouts would be set at the same time, but since the second is executed inside the first it only needs 60 sec

PR #2495 (2023-01-13): Health - Fix heartrate sensor timeout
Fixes #2435
closed 2023-01-13 by gfwilliams