espruino/BangleApps #2460: [messages] Messages seems to not save sometimes

Kedlub, 2023-01-06:
### Affected hardware version

Bangle 2

### Your firmware version

2v16

### The bug

When going back to the clock or normal loading an app loses recently received messages, even when clicking the message widget while in an app.

### Expected behavior
Recently received messages don't disappear while loading an app

### Steps to reproduce
1. Open any app
2. Receive notification
3. Click on the notification widget, or return to the clock
4. Observe that the notification is gone

### Installed apps

health, setting, welcome, antonclk, hidcam, info, locale, widpedom, scicalc, run, calendar, henkinen, slidingtext, fuzzyw, stlap, mylocation, chimer, authentiwatch, widminbt, kbmulti, banglexercise, about, widalarm, sleeplog, noteify, files, hrm, sleepphasealarm, boot, rebble, clkinfosunrise, openstmap, recorder, torch, lato, astrocalc, messageicons, messages, presentor, ios, widmessages, choozi, widlock, wid_edit, messagegui, fastload, alarm, qmsched, sched, widclk, iconlaunch, widbars, barclock, widChargingStatus, gpsautotime, swiperclocklaunch

Kedlub, 2023-02-08:
Retested again on a clean watch with system 2v16, the only apps were: boot (0.55), launch (0.20), antonclk (0.11), about (0.14), notify (0.12), health (0.22), widbatv (0.01), setting (0.56), messageicons (0.04), messagegui (0.64), widmessages (0.04), messages (0.58), widbt (0.08), android (0.23)

## Steps to reproduce
0. Reinstall default apps
1. Install iOS or Android integration
2. Enter any app (for example About)
3. Receive a notification (icon is flashing in widget area)
4. Exit the app
5. Notification is gone (both in the widget area and in the messages list)

Kedlub, 2023-02-09:
Tried to exit both using app back button and by a long press of the hardware button. It happens on every normal load (the one that shows "Loading..." on the screen. I think it started happening after change #2373, but that is weird because it should save messages when exiting an app

gfwilliams, 2023-02-10:
Thanks - I'm afraid I'm off next week, but I'll make a note to look at this when I get back

dapgo, 2023-02-22:
I am seeing the same on Bangle.js1, I see notification on the  widget but most times I have "No messages" on in messages app.

gfwilliams, 2023-02-23:
Ok, thanks. I just tried this and can confirm as well.

Sending `GB({"t":"notify","id":1575479849,"src":"WhatsApp","title":"A Name","body":"message contents"})` from the IDE when in the About app causes the following error, which I guess is why it never gets saved:

```
Uncaught ReferenceError: "event" is not defined
 at line 33 col 44 in messagegui
...es = (Bangle.CLOCK || event.important); // should we load th...
                              ^
in function "listener" called from line 18 col 77 in .boot0
require("messagegui").listener(type, msg)
                                        ^
in function called from system
```

Just fixed

closed 2023-02-23 by gfwilliams

lunctis-viribus, 2023-03-01:
Hi @gfwilliams @Kedlub,
Is it possible that after the update of Message UI (v0.64->v0.65) when having the iconlaunch app, when (auto)closing a message the iconlaunch app is opened instead of the default clock app? Hope this gets fixed. Thanks.

gfwilliams, 2023-03-01:
What happens if you run `load()` in the Web IDE? That *should* load the clock, but if it loads `iconlaunch` then I guess that would be the problem.

Do you press a button to close the message? I guess it's possible that the message UI closed when the button is pressed but the launcher opens when it is released, so the two things happen quite quick

lunctis-viribus, 2023-03-01:
`load()` in Web IDE does load `iconlaunch`.
Is this then an issue of `iconlaunch` or my clock app `weatherClock` (0.05)? I didn't had this issue before yesterday when I updated 4 apps, one of which was `Message UI` and not including `iconlaunch`.

gfwilliams, 2023-03-01:
Ok, I think somehow your watch has got very confused then! Can you try running these commands in the Web IDE:

```
var s = require("Storage").readJSON("setting.json",1)||{};
print(s.clock);
```

And also:

```
print(require("Storage").list(/\.info$/)
    .map(file => {
      const app = require("Storage").readJSON(file,1);
      if (app && app.type == "clock") {
        return app;
      }
    })
    .filter(x=>x)
    .sort((a, b) => a.sortorder - b.sortorder)[0]);
```

And tell me the result?

lunctis-viribus, 2023-03-01:
Output 1:
weatherClock.app.js
Output 2:
{
  "id": "antonclk",
  "name": "Anton Clock",
  "type": "clock",
  "src": "antonclk.app.js",
  "icon": "antonclk.img",
  "version": "0.11",
  "tags": "clock",
  "files": "antonclk.info,antonclk.app.js,antonclk.img"
 }
Output with argument [1] instead of [0]:
{
  "id": "weatherClock",
  "name": "Weather Clock",
  "type": "clock",
  "src": "weatherClock.app.js",
  "icon": "weatherClock.img",
  "version": "0.05",
  "tags": "clock, weather",
  "files": "weatherClock.info,weatherClock.app.js,weatherClock.img"
 }

gfwilliams, 2023-03-01:
Thanks - and what happens if you do `load("weatherClock.app.js")`?

lunctis-viribus, 2023-03-01:
That gives the weatherClock screen, similar to long press button (but without 'Loading...' overlay).

lunctis-viribus, 2023-03-01:
The invalid Messages Settings is resolved by reinstalling without the option 'Minify apps before upload'.

`iconlaunch` is still loaded instead of `weatherClock`...

gfwilliams, 2023-03-01:
> The invalid Messages Settings is resolved by reinstalling without the option 'Minify apps before upload'.

Oh right. Please can you ensure that's off and click `reinstall all apps`? That's probably the cause of all your problems - there's a good reason it says:

>BETA, not recommended. Uploads smaller, faster apps but this may cause some apps to stop working

lunctis-viribus, 2023-03-01:
I twice reinstalled all apps. Issue is the same. Can it be caused by this change or should I make a separate bug issue?

gfwilliams, 2023-03-01:
Well, right now I'd be pretty sure it's you custom clock app. I tried here and it all works fine for me with the apps in the app store.

So just to be 100% clear, if you:

* Receive a message (or eg type `GB({"t":"notify","id":1575479849,"src":"Hangouts","title":"A Name","body":"message contents"})` in the IDE to simulate getting one
* Press the 'X' button on the message to exit and go back to the clock

You go back to the launcher instead?

I just checked and pressing X in the messages app calls 'load()' which we tested goes back to the clock app fine, so I'm not sure what's going on here

lunctis-viribus, 2023-03-01:
It's solved. I have installed the app `fastload` and accidentally enabled the option 'Force load to launcher' in Settings... (the settings app sometimes acts weird when clicking the upper-left back button widget). Thanks for the help!

commit 8bc9c5de (2023-02-23): messagegui 0.65: Make sure messages are saved if not in the clock app (fix #2460)