espruino/BangleApps #2468: circlesclock: HRM always enabled

gellnerm, 2023-01-07:
If the HRM circle is selected, the HRM is never stopped, even though the watch display is off and locked. This causes battery drain in the long run.
I think the heart rate measured by the Health Tracking app should be used instead if possible. It measures HR in the background every n minutes.

4 circles are selected.
LCD timeout is set to 30s. 

Please let me know if further information is required.

myxor, 2023-01-08:
Just for info:
I don't think this comes from Circlesclock but from clock_info module: https://github.com/espruino/BangleApps/blob/059a2e7af70f67d115da02b3d70b37626bb0fec6/modules/clock_info.js#L113

gfwilliams, 2023-01-09:
Yes - this was intentional though... I figured if you wanted the heart rate displayed, you probably wanted to see your current heart rate?

I feel like if you turned to display the heart rate and only your heart rate from the last 10 minutes (*if* it had been enabled) was shown, I'd probably get more complaints?

gellnerm, 2023-01-09:
Sounds compelling, it's a question of personal taste. Maybe it could be two separate options that can be selected.

However the "doesn't stop measuring even though display off" issue remains.

thyttan, 2023-01-09:
>However the "doesn't stop measuring even though display off" issue remains.

It can be argued that if the heart rate clock info is displayed it should always keep the measurement up to date even if the screen is locked. The middle ground, that you maybe are proposing, could be to have heart rate continually updated when the screen is unlocked, but fall back to using the update interval set in 'Health' settings when the screen is locked? I personally don't have a strong opinion on this.

gfwilliams, 2023-01-09:
> However the "doesn't stop measuring even though display off" issue remains.

Are you talking about Bangle.js 1?

I can see it there - because the HRM draws a lot more power *and* the screen goes off completely.

I think at some point soon we'll move the ClockInfo code to its own app (in the app loader), and then we can have a settings page where stuff like this can be configured

gellnerm, 2023-01-09:
> have heart rate continually updated when the screen is unlocked, but fall back to using the update interval set in 'Health' settings when the screen is locked

This would be a good compromise. Personally, I'd like to see only the "background 10s HR", because

1. I want to have the lowest possible battery consumption and active HR costs 1.0mA according to the [docs](https://www.espruino.com/Bangle.js2#power-consumption)
2. I don't want to wait for accurate HR values, because this takes ~30s (my observation) and I would have to increase LCD timeout while the background HR seems accurate at first glance (I guess it also measures ~30s long)

At the moment I switched to the LCARS clock, where background HR is already used.

> Are you talking about Bangle.js 1?

It's the Bangle.js 2.

> a settings page where stuff like this can be configured

I can perfectly live with this being planned in the future. You may want to decide to close this ticket, thanks.

gfwilliams, 2023-01-09:
Ok, great. Might as well leave this open until the option gets added.

Just a note though:

> I'd like to see only the "background 10s HR"

This wouldn't be 10 second - it'd be 10 *minute*. Is that  a problem?

bobrippling, 2023-11-15:
This was fixed when `clock_info` was moved into a module, de40f314c7037b.

Here we can see the code turns the HRM off and then relies on the usual health HRM events:

https://github.com/espruino/BangleApps/blob/865ca1f518bc39d86780537775b4106873a2a9fd/apps/clock_info/lib.js#L92-L102

closed 2023-11-15 by bobrippling