espruino/BangleApps #2474: Hideable widget bar blank when first shown

jt-nti, 2023-01-08:
### Affected hardware version

Bangle 2

### Your firmware version

2v16

### The bug

When swiping down the hidden widget bar I see a blank bar with no widgets visible

I've tried with the MacWatch2 clock and the widget bar seems to stay blank until after the screen locks/unlocks. After that the widgets (lock widget, lato pedometer, and battery widget) are shown as expected when swiping to reveal the hidden widget bar.

### Installed apps

widbat (0.09), waypoints (0.01), cprassist (0.02), widpb (0.05), calendar (0.11), antonclk (0.11), gpstime (0.05), beer (0.02), quicklaunch (0.08), boot (0.55), messagegui (0.63), barcode (0.10), messages (0.58), messageicons (0.03), launch (0.20), widalarm (0.01), about (0.14), lato (0.01), medicalinfo (0.01), geminiclock (0.01), gpsinfo (0.11), alarm (0.37), widmessages (0.04), widlock (0.08), setting (0.56), sched (0.18), notify (0.12), health (0.19), android (0.20), widclkbttm (0.05), macwatch2 (0.06)

sir-indy, 2023-01-17:
The above commit seems to have fixed this issue now, Gordon found the solution far quicker than I could!

sir-indy, 2023-01-17:
Yes, sorry, this only fixes the issue of a blank widget bar when first shown. I haven't tried installing your widmeda, though it looks very good!

closed 2023-01-18 by gfwilliams