espruino/BangleApps #2477: [BW Clock] Can't open launcher correctly when plugging in charger...

thyttan, 2023-01-09:
### Affected hardware version

Bangle 2

### Your firmware version

2v16.49

### The bug

... except if the battery clock info was already shown prior to connecting the charger.

Reproduce:
1. Have BW Clock open
2. make sure to not be on the battery clock info
3. connect charger
4. battery clock info is automatically displayed
5. press hardware button
6. The UI stops responding to screen or hardware button input
7. (after a second or two the  analog clock widget is displayed if installed)
8. (Long press HW button to trigger `load()`ing clock anew, which works)

I think this might be what @BartS23 points out in https://github.com/espruino/BangleApps/pull/2471#discussion_r1064500843

### Installed apps

messagelist, health, bwclk, dtlaunch, sched, kbmulti, widbthide, widram, calculator, activityreminder, pomoplus, widanclk, kbtouch, run, heartzone, wohrm, sevenmin, banglexercise, sleeplog, swp2clk, torch, hrm, lightswitch, boot, recorder, swscroll, spotrem, podadrem, sleepphasealarm, annoy, messages, widmsggrid, messageicons, messagesmusic, slopeclock, fastload, tetris, smpltmr, weather, circlesclock, alarm, android, setting, dragboard, aiclock, widminbat, notify, quicklaunch, imageclock, taglaunch, altimeter.

All updated just recently via development app loader.

gfwilliams, 2023-01-09:
Is this definitely with the 'new' BW Clock that uses 'addInteractive'?

thyttan, 2023-01-09:
>Is this definitely with the 'new' BW Clock that uses 'addInteractive'?

Yes

thyttan, 2023-01-09:
Small addition below:

Reproduce:
1. Have BW Clock open
2. make sure to not be on the battery clock info
3. connect charger
4. battery clock info is automatically displayed
5. press hardware button
6. The UI stops responding to screen or hardware button input
7. (after a second or two the  analog clock widget is displayed if installed) **(the bottom part of BW Clock will also redraw so the "Fastloading..." message is partially hidden)**
8. (Long press HW button to trigger `load()`ing clock anew, which works)

thyttan, 2023-01-09:
Debug output from Web IDE, sorry for not adding it to begin with.

```
>
{"t":"status","bat":89,"chg":0}
Uncaught Error: clearInterval(undefined) not allowed. Use clearInterval() instead.
 at line 92 col 54 in clock_info
clearInterval(this.interval); delete this.interval; Bangle.r...
                           ^
in function "hide" called from line 227 col 16 in clock_info
      itm.hide();
               ^
in function "menuHideItem" called from line 289 col 58 in clock_info
    menuHideItem(menu[options.menuA].items[options.menuB]);
                                                         ^
in function "remove" called from line 299 col 24 in bwclk.app.js
clockInfoMenu.remove();
                     ^
in function "kill" called from line 324 col 10 in bwclk.app.js
    kill();
         ^
in function "a" called from line 2 col 168
...e;delete Bangle.uiRemove;a()}g.reset();if(c){if("updown"==c)...
                              ^
in function called from line 1 col 31
Bangle.uiRemove?(Bangle.setUI(),(__FILE__=a)||(a=".bootcde")...
                              ^
at line 1 col 115
..."Storage").read(a))):load(a)
                              ^
in function called from line 75 col 11 in .boot0
    o(name);
          ^
at line 75 col 12 in .boot0
    o(name);
           ^
in function "o" called from line 902 col 8 in .boot0
o(f);
   ^
in function "load" called from line 1 col 408
...))}c?Bangle.load(b.launcher):eval('E.showMessage("No Launche...
                              ^
in function called from system
Uncaught Error: clearInterval(undefined) not allowed. Use clearInterval() instead.
 at line 92 col 54 in clock_info
clearInterval(this.interval); delete this.interval; Bangle.r...
                           ^
in function "hide" called from line 227 col 16 in clock_info
      itm.hide();
               ^
in function "menuHideItem" called from line 289 col 58 in clock_info
    menuHideItem(menu[options.menuA].items[options.menuB]);
                                                         ^
in function "remove" called from line 299 col 24 in bwclk.app.js
clockInfoMenu.remove();
                     ^
in function called from system
>
{"t":"status","bat":89,"chg":1}
Uncaught Error: clearInterval(undefined) not allowed. Use clearInterval() instead.
 at line 92 col 54 in clock_info
clearInterval(this.interval); delete this.interval; Bangle.r...
                           ^
in function "hide" called from line 227 col 16 in clock_info
      itm.hide();
               ^
in function "menuHideItem" called from line 289 col 58 in clock_info
    menuHideItem(menu[options.menuA].items[options.menuB]);
                                                         ^
in function "remove" called from line 299 col 24 in bwclk.app.js
clockInfoMenu.remove();
                     ^
in function "kill" called from line 324 col 10 in bwclk.app.js
    kill();
         ^
in function "a" called from line 2 col 168
...e;delete Bangle.uiRemove;a()}g.reset();if(c){if("updown"==c)...
                              ^
in function called from line 1 col 31
Bangle.uiRemove?(Bangle.setUI(),(__FILE__=a)||(a=".bootcde")...
                              ^
at line 1 col 115
..."Storage").read(a))):load(a)
                              ^
in function called from line 75 col 11 in .boot0
    o(name);
          ^
at line 75 col 12 in .boot0
    o(name);
           ^
in function "o" called from line 902 col 8 in .boot0
o(f);
   ^
in function "load" called from line 1 col 408
...))}c?Bangle.load(b.launcher):eval('E.showMessage("No Launche...
                              ^
in function called from system
> 
```

PR #2480 (2023-01-09): Update clock_info, fix [BW clock]
Fixes #2477

pinging @thyttan and @peerdavid
closed 2023-01-09 by gfwilliams