espruino/BangleApps #2478: [widget utils/Fastload Utils] Initial .hide() will not hide widgets when fast loading, needs a .show() first

thyttan, 2023-01-09:
### Affected hardware version

Bangle 2

### Your firmware version

2v16.49

### The bug

First discussed in https://github.com/espruino/BangleApps/pull/2464#issuecomment-1375354642.

I experience this with Podcast Addict Remote and Remote for Spotify, when fast loading from BW Clock and from Desktop Launcher (both showing widgets) using Fastload Utils installed on the watch. The bug is not present if not using Fastload Utils.

I have yet to see if this bug is present when switching to other apps than the two mentioned using .hide(). I have also not tried launching them from other clocks or launchers.

Reproduce:
1. Install Desktop Launcher and Fastload Utils
2. open Podcast Addict Remote
3. Notice widgets are not hidden.

### Installed apps

kbmulti, widbthide, widram, calculator, activityreminder, pomoplus, widanclk, kbtouch, run, heartzone, wohrm, sevenmin, banglexercise, sleeplog, swp2clk, torch, hrm, lightswitch, boot, recorder, swscroll, spotrem, podadrem, sleepphasealarm, annoy, messages, widmsggrid, messageicons, messagesmusic, slopeclock, fastload, tetris, smpltmr, weather, circlesclock, alarm, android, setting, dragboard, aiclock, widminbat, notify, quicklaunch, taglaunch, altimeter, messagelist, health, bwclk, dtlaunch, sched, widalarm

thyttan, 2023-01-09:
```
/*
1. Upload this script to a file "testWU.js" via Web IDE.
2. Navigate the watch to a clock/launcher/app with remove-function specified and widgets shown (I use BW Clock).
3. In console field enter "Bangle.load('testWU.js')".
4. Notice if widgets are hidden or not (Most of the time they would not be hidden for me and be interactable ,e.g. Light Switch Widget, the red back button will be hidden but interactable).
5. Click hardware button to load the user chosen clock (for the following steps it should have a remove function, or specify another app in the back function below)
6. Notice if widgets are shown or not (On BW Clock they are shown as expected).
7. In console field again enter "Bangle.load('testWU.js')".
8. Notice if the widgets are hidden or not (They are not hidden for me).
9. In console field again enter "Bangle.load('testWU.js')".
10. Notice if the widget are hidden or not (This time they are hidden for me, however if I tap upper left corner the back widget still executes and if I tap the upper right corner Light Switch Widget will be triggered).
11. Press hardware button to load the user chosen clock.
12. Notice if widgets are shown (they are shown for me).

The expected behavior would of course be that the widgets are always hidden when testWU.js is launched.
*/

{
g.clear();

  Bangle.setUI({
    mode : "custom",
    back : ()=>{Bangle.load();},
    remove : ()=>{require('widget_utils').show();}
  });

  Bangle.loadWidgets();
  require('widget_utils').hide();
}
```

gfwilliams, 2023-01-10:
@halemmerich any ideas about this one?

halemmerich, 2023-01-11:
@thyttan I have tried your steps with BWClock and I see the same behaviour there.
Tests with Anton and newly fastloadable Berlin Clock do not show the problem.

So at first glance I would suspect a problem in BWClock. It does not seem to use widget_utils for hiding and showing the widgets so that is probably the cause or at least part of it.

thyttan, 2023-01-11:
I'll try adding widget utils in BW Clock then, then we can check it off on #2196 as well. I looked at doing that before but didn't really understand the current handling of widgets though 🙃 Maybe I'll get it this time.

thyttan, 2023-01-11:
So when I add widget utils to BW Clock .hide() works directly with the test code above as well as the two aforementioned apps.

However, interactable widgets such as the back button (via setUI back function) and Light Switch Widget will still be triggered if I touch where they usually are.

thyttan, 2023-01-11:
Aha! :)

thyttan, 2023-01-11:
I don't know if something should be done so .hide() works even in the case when loading from a clock app which doesn't use widget_utils? Otherwise it's ok with me to close this.

#2492 Converts all remaining clocks to use widget_utils.

halemmerich, 2023-01-11:
I don't think that is easily doable. The problem was BWClock attempting the same things as widget_utils (renaming and changing `draw`/`_draw`). So essentially converting everything to widget_utils and having an eye on future changes trying to do something like it should solve this.

thyttan, 2023-01-11:
Thank you both for checking this out!

closed 2023-01-11 by thyttan

gfwilliams, 2023-01-12:
> I think that is expected since hide/show currently only handles the drawing code, not the listeners of the loaded widgets.

Yes, it is. Not ideal but I think for most cases it's not an issue.

> I don't know if something should be done so .hide() in an app works even in the case when loading from a clock app which doesn't use widget_utils? 

Fast load should only work for apps that have 'opted in' by adding `remove` listener - so really if an app has added 'fast load' and it's breaking other stuff because of something like its widget hiding, that means that whoever added it didn't test it properly.