espruino/BangleApps #2485: [weather] Weather Widget Always Shows Celcius Value

atomdmac, 2023-01-09:
### Affected hardware version

Bangle 2

### Your firmware version

2v12

### The bug

### Description
I'm using the OpenWeather app as the data provider for the `weather` app on my BangleJS 2.  The widget appears to always show the temperature in Celsius units (even if the app on my phone is displaying them in Fahrenheit).

### Installed apps

widbat (0.09), widminbt (0.01), wdstepcstm (0.01), mylocation (0.09), antonclk (0.11), boot (0.55), messageicons (0.03), messages (0.58), swscroll (0.02), launch (0.20), quicklaunch (0.08), messagegui (0.63), notify (0.12), health (0.19), widlock (0.08), weather (0.23), setting (0.56), android (0.20), entonclk (0.2), widmessages (0.04)

gfwilliams, 2023-01-10:
I notice you don't have the  `Languages` app installed: https://banglejs.com/apps/?id=locale

Please could you try installing that with it set to your country, and it should *hopefully* convert temperature to F for you. 

If that doesn't fix it then some extra changes might be needed though

atomdmac, 2023-01-11:
@gfwilliams it looks like that was the step that I was missing. I set my locale to US and it's working perfectly now. Thank you very much for your help!

gfwilliams, 2023-01-12:
Ok, great!

closed 2023-01-12 by gfwilliams