espruino/BangleApps #2489: [android/messages] Message Tick/Cross button implementation

gfwilliams, 2023-01-11:
### Affected hardware version

Bangle 2

### Your firmware version

2v16

### The bug

When a message comes in on Android, you often have two buttons, Tick/Cross.

On iPhone, these are positive and negative responses to a message if available (according to Apple).

Currently,

Tick -> 'OPEN' in Gadgetbridge
Cross -> 'DISMISS' in Gadgetbridge

These are from `GBDeviceEventNotificationControl.Event` which can have the following types:

```
        DISMISS,
        DISMISS_ALL,
        OPEN,
        MUTE,
        REPLY,
```

Right now, as far as I can tell, OPEN will open the notification in Android, and DISMISS actually removes it from the device

So what should the two buttons actually do? 'OPEN' doesn't seem useful since the device is in your pocket.

Should Tick call DISMISS, and Cross just mark the message read on Bangle.js and do nothing else? Or do we just have a single button that calls DISMISS, and 'back' marks read anyway?

... and when we get this figured out, we'll probably just get rid of the buttons and move to a swipe in either direction - giving more screen real estate for the text

### Installed apps

android,messages

pebl-hank, 2023-01-18:
I'd be happy to have a check mark (or an eye icon) that marks the element as read on the phone and another button with some kind of do not disturb icon (crossed loudspeaker or so) or a cross that just dismisses it on the watch.
Also the watch app should present an option to mark everything unread on the phone and watch as read. And an option to show all unread messages in a list.

thyttan, 2023-05-22:
I would probably like for one of the buttons to trigger a thumbs up reply to the message.

I'm also thinking maybe one of the buttons (or swipes if we move to that) could lead to some number of predefined replies - either text or emojis. Perhaps configurable by the user somehow (via the app loader seems like a good idea to me).

I think @pebl-hank's idea sounds sensible also.

jfrederickson, 2023-05-22:
One thing I'd really like from a message action implementation is to be able to interact with Android's actionable notifications. I use a two-factor auth app for work that sends me a notification with allow/deny actions, and currently I have to pull out my phone to respond to those. Would be very convenient if I could just allow the login from my watch!

gfwilliams, 2023-05-23:
I believe we do have some support for actionable notifications in Gadgetbridge, passed to Bangle.js - so potentially something could be done there. Also I believe replies are possible too.

But honestly, changing the 'tick' to respond with thumbsup would be the wrong option for most people I think - of course I guess that could be configurable in the `Android` app for Bangle.js if someone implemented it...

pebl-hank, 2023-05-23:
I like the idea of having a general reply button that is optional configurable (show/don't show) in the settings. That would lead to like 5 or 10 custom messages a user can configure in Gadgetbridge or in the Bangle App loader.